motioneye-project/motioneye #2494: How to record remote motionEye Camera in motionEye hub?

wecoon, 2022-05-20:
Hi everyone, 

I got two Raspberries to begin my setup. On both, I've installed Rasbian and motionEye webserver. 

Pi Zero has a connected camera.
Pi 4 is working as a hub and has the Pi Zero camera connected as a "remote motionEye Camera"

I have bigger SD card on Pi 4 as I planned to add more Pi Zero cameras and record only on the hub. However, on the hub settings, I see only camera device settings, like Pi Zero SD card as storage. 

Screenshot from the hub, showing the camera's (Pi Zero) storage:
[image]

**How could I record the "remote motionEye Camera" video stream on the hub?**

Some additional info on my attempts:
I noticed, that if I add the camera as "Network Camera", then the settings show the hub's storage. Sadly, in this case, the camera is not working. It does stream if I set it as a "Simple MJPEG Camera", but then there are no storage or recording settings. In both cases, I use the same URL and port 8081.

from left to right: "remote motionEye Camera", "Simple MJPEG Camera", "Network Camera"
[image]

wecoon, 2022-05-20:
OK, so I worked around the issue. My Hub is running Raspbian with installed newest webserver version, and the camera is running the latest stable motionEyeOS. It is connected as a network camera, latency and quality are just amazing compared to what I had before. I guess I should build a new version to OS somehow, but I am new to the raspberry world, so this will gonna work for now. Any suggestions, tips and recommendations are welcome. 

I guess this could be closed, but there are issues with the latest solution, and I saw other people reporting something similar.

MichaIng, 2022-05-21:
In case of "Network Camera", could you show the output of `journalctl -u motioneye`?

Generally "Network Camera" seems to be the most reasonable choice if you want to handle and record everything on the RPi 4/hub. If "remote motionEye Camera" would do just the same, there would be no reason to make this different options:
- "remote motionEye Camera": stream handling, motion detection, recording all done on remote motionEye server, only frontend on the hub.
- "Network Camera": Everything done on the motionEye hub.
- "Simple MJPEG Camera": Direct pass-through of the remote camera stream to the client, no handling at any motionEye server(?)

As far as I understood so far 😅.

wecoon, 2022-05-22:
@MichaIng thanks for the clarification. This makes sense. 

I had issues with buster and motionEye on Pi Zero, now it is running old 202006 motionEyeOs and it all works perfectly as a Network camera. Hub is still with bullseye and motionEye webserver. I don't have issues, so I cannot give you logs anymore :)

MichaIng, 2022-05-22:
I wonder whether it has to do with #2486.

wecoon, 2022-05-22:
Sorry, do not know. I can only tell, that I did not change auth, admin without password was used

JamBalaya56562, 2026-11-10:
Closing with the answer: a Remote motionEye Camera is recorded by the Motion on the Pi Zero, so for the hub to record, add the Pi Zero's stream (`http://<pi-zero>:8081/`) as a Network Camera instead; the hub's Motion then detects and records it to the hub's storage. The stream authentication fixes of #2486, in every release since 0.43.1, were what that Network Camera was likely missing here.

closed 2026-11-10 by JamBalaya56562