motioneye-project/motioneye #2495: RPi Zero 2 W - install issue - python setup.py egg_info did not run successfully

khaiin, 2022-05-21:
Hi,
I've started a fresh installation on RPi Zero 2W following the installation instructions - [https://github.com/motioneye-project/motioneye/tree/dev](https://github.com/motioneye-project/motioneye/tree/dev) . At point 3. Install and setup motionEye I have gotten stuck on the error below. Any suggestion if there is something on my side I've missed, or can do to  continue?
Thanks.

```
user@camone:~ $ sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting https://github.com/motioneye-project/motioneye/archive/dev.tar.gz
  Downloading https://github.com/motioneye-project/motioneye/archive/dev.tar.gz
     - 1.0 MB 4.1 MB/s 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting babel
  Downloading https://www.piwheels.org/simple/babel/Babel-2.10.1-py3-none-any.whl (9.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 9.5/9.5 MB 3.5 MB/s eta 0:00:00
Collecting pillow
  Downloading Pillow-9.1.1-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (3.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.0/3.0 MB 3.8 MB/s eta 0:00:00
Collecting tornado
  Downloading tornado-6.1-cp39-cp39-manylinux2014_aarch64.whl (427 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 427.7/427.7 kB 4.3 MB/s eta 0:00:00
Collecting boto3
  Downloading https://www.piwheels.org/simple/boto3/boto3-1.23.5-py3-none-any.whl (129 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 129.8/129.8 kB 1.1 MB/s eta 0:00:00
Collecting jinja2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 kB 1.6 MB/s eta 0:00:00
Collecting pycurl
  Downloading pycurl-7.45.1.tar.gz (233 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 233.9/233.9 kB 2.2 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [27 lines of output]
      /tmp/pip-install-ydil_4_l/pycurl_9fff5f578d214041824d8281395dcf3a/setup.py:957: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
        if LooseVersion(distutils.__version__) > LooseVersion("1.0.1"):
      /tmp/pip-install-ydil_4_l/pycurl_9fff5f578d214041824d8281395dcf3a/setup.py:959: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
        if LooseVersion(distutils.__version__) < LooseVersion("1.0.3"):
      Traceback (most recent call last):
        File "/tmp/pip-install-ydil_4_l/pycurl_9fff5f578d214041824d8281395dcf3a/setup.py", line 235, in configure_unix
          p = subprocess.Popen((self.curl_config(), '--version'),
        File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
          self._execute_child(args, executable, preexec_fn, close_fds,
        File "/usr/lib/python3.9/subprocess.py", line 1823, in _execute_child
          raise child_exception_type(errno_num, err_msg, err_filename)
      FileNotFoundError: [Errno 2] No such file or directory: 'curl-config'
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-ydil_4_l/pycurl_9fff5f578d214041824d8281395dcf3a/setup.py", line 1017, in <module>
          ext = get_extension(sys.argv, split_extension_source=split_extension_source)
        File "/tmp/pip-install-ydil_4_l/pycurl_9fff5f578d214041824d8281395dcf3a/setup.py", line 673, in get_extension
          ext_config = ExtensionConfiguration(argv)
        File "/tmp/pip-install-ydil_4_l/pycurl_9fff5f578d214041824d8281395dcf3a/setup.py", line 99, in __init__
          self.configure()
        File "/tmp/pip-install-ydil_4_l/pycurl_9fff5f578d214041824d8281395dcf3a/setup.py", line 240, in configure_unix
          raise ConfigurationError(msg)
      __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory: 'curl-config'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

MichaIng, 2022-05-21:
> I'm on a Raspbian Lite 64-bit Bullseye

> Downloading https://www.piwheels.org/

piwheels does not support 64-bit. At best remove the installed pre-compiled wheels/modules from piwheels and install motionEye again with the instructions for "other" systems (not ARMv6/ARMv7).

Tibladar, 2022-05-21:
> No such file or directory: 'curl-config'

Just install the dependencies, no need to remove any pi modules

`sudo apt install libcurl4-openssl-dev libssl-dev`

MichaIng, 2022-05-21:
> Just install the dependencies, no need to remove any pi modules

The modules however are compiled against ARMv6, and pip configured via `/etc/pip.conf` to pull such from piwheels. IMO better to run native aarch64 wheels on aarch64 systems.

khaiin, 2022-05-22:
> > No such file or directory: 'curl-config'
> 
> Just install the dependencies, no need to remove any pi modules
> 
> `sudo apt install libcurl4-openssl-dev libssl-dev`

Hi,
tried to bypass it by installing dependencies and it created another error:

```
user@camone:~ $ sudo apt install libcurl4-openssl-dev libssl-dev
... done OK.
user@camone:~ $ sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting https://github.com/motioneye-project/motioneye/archive/dev.tar.gz
  Downloading https://github.com/motioneye-project/motioneye/archive/dev.tar.gz
     - 1.1 MB 3.8 MB/s 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting boto3
  Using cached https://www.piwheels.org/simple/boto3/boto3-1.23.5-py3-none-any.whl (129 kB)
Collecting babel
  Using cached https://www.piwheels.org/simple/babel/Babel-2.10.1-py3-none-any.whl (9.5 MB)
Collecting tornado
  Using cached tornado-6.1-cp39-cp39-manylinux2014_aarch64.whl (427 kB)
Collecting pillow
  Using cached Pillow-9.1.1-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (3.0 MB)
Collecting pycurl
  Using cached pycurl-7.45.1.tar.gz (233 kB)
  Preparing metadata (setup.py) ... done
Collecting jinja2
  Using cached https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting pytz>=2015.7
  Downloading https://www.piwheels.org/simple/pytz/pytz-2022.1-py3-none-any.whl (511 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 511.2/511.2 kB 2.1 MB/s eta 0:00:00
Collecting s3transfer<0.6.0,>=0.5.0
  Downloading https://www.piwheels.org/simple/s3transfer/s3transfer-0.5.2-py3-none-any.whl (79 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 79.5/79.5 kB 1.4 MB/s eta 0:00:00
Collecting jmespath<2.0.0,>=0.7.1
  Downloading https://www.piwheels.org/simple/jmespath/jmespath-1.0.0-py3-none-any.whl (20 kB)
Collecting botocore<1.27.0,>=1.26.5
  Downloading https://www.piwheels.org/simple/botocore/botocore-1.26.5-py3-none-any.whl (8.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.8/8.8 MB 2.9 MB/s eta 0:00:00
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.1-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (26 kB)
Requirement already satisfied: urllib3<1.27,>=1.25.4 in /usr/lib/python3/dist-packages (from botocore<1.27.0,>=1.26.5->boto3->motioneye==0.43.0) (1.26.5)
Collecting python-dateutil<3.0.0,>=2.1
  Downloading https://www.piwheels.org/simple/python-dateutil/python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 247.7/247.7 kB 1.4 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil<3.0.0,>=2.1->botocore<1.27.0,>=1.26.5->boto3->motioneye==0.43.0) (1.16.0)
Building wheels for collected packages: motioneye, pycurl
  Building wheel for motioneye (pyproject.toml) ... done
  Created wheel for motioneye: filename=motioneye-0.43.0-py3-none-any.whl size=781808 sha256=9dda84ac2e201ffe6b7f3c4bb23dfe20acd20341a36c99ef292a6cc5bf9d5f03
  Stored in directory: /tmp/pip-ephem-wheel-cache-4m52fokl/wheels/5d/19/3b/8df45c45f762bba3d414e214316b57965795dbbf58c209bce5
  Building wheel for pycurl (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [24 lines of output]
      /tmp/pip-install-dpahx1sg/pycurl_5fd6aef4a95447208b9313d783b3dc62/setup.py:957: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
        if LooseVersion(distutils.__version__) > LooseVersion("1.0.1"):
      /tmp/pip-install-dpahx1sg/pycurl_5fd6aef4a95447208b9313d783b3dc62/setup.py:959: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
        if LooseVersion(distutils.__version__) < LooseVersion("1.0.3"):
      Using curl-config (libcurl 7.74.0)
      Using SSL library: OpenSSL/LibreSSL/BoringSSL
      running bdist_wheel
      running build
      running build_py
[... 41 lines cut ...]
      creating build/lib.linux-aarch64-cpython-39
      creating build/lib.linux-aarch64-cpython-39/curl
      copying python/curl/__init__.py -> build/lib.linux-aarch64-cpython-39/curl
      running build_ext
      building 'pycurl' extension
      creating build/temp.linux-aarch64-cpython-39
      creating build/temp.linux-aarch64-cpython-39/src
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DPYCURL_VERSION=\"7.45.1\" -DHAVE_CURL_SSL=1 -DHAVE_CURL_OPENSSL=1 -DHAVE_CURL_SSL=1 -I/usr/include/python3.9 -c src/docstrings.c -o build/temp.linux-aarch64-cpython-39/src/docstrings.o
      In file included from src/docstrings.c:4:
      src/pycurl.h:5:10: fatal error: Python.h: No such file or directory
          5 | #include <Python.h>
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pycurl

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

Tibladar, 2022-05-22:
Have you run the apt install instructions for `other architectures` in https://github.com/motioneye-project/motioneye/tree/dev#installation ?

`fatal error: Python.h: No such file or directory`
Probably missing python3-dev
`sudo apt install python3 python3-dev`

khaiin, 2022-05-22:
> Have you run the apt install instructions for `other architectures` in https://github.com/motioneye-project/motioneye/tree/dev#installation ?
> 
> `fatal error: Python.h: No such file or directory` Probably missing python3-dev `sudo apt install python3 python3-dev`

OK, so it was that little guy - `python3-dev`, who was missing. After manual apt installation, the installation of motionEye completed successfully. So both of you were right to follow the "other system" instructions just in other words.

Meanwhile on another SD card, I've created another installation of 64-bit Raspbian Lite with the instructions as "Other" system, and went through successfully as well, following only commands:
```
sudo apt update
sudo apt --no-install-recommends install ca-certificates curl python3 python3-dev libcurl4-openssl-dev gcc libssl-dev
curl -sSfO 'https://bootstrap.pypa.io/get-pip.py'
sudo python3 get-pip.py
rm get-pip.py
sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
sudo motioneye_init
```
Also earlier today I've successfully completed the installation of motionEye on 32-bit Raspbian Lite based on the ARMv6/ARMv7 instructions.

So not sure it it is possible to generalize, but would it make sense to update the instructions and instead of  `On ARMv6 and ARMv7 systems` write `On 32-bit ARMv6 and ARMv7 systems` ? So newbs like me can follow it without hustle and avoid this discussion we've had :] ?

@MichaIng and @Tibladar, big thank you for taking your time and helping me start the journey with the motionEye software.
Cheers

MichaIng, 2022-05-22:
ARMv6/ARMv7 basically implies 32-bit, since those CPU architectures do not support 64-bit 😉. But yes, we can add the, of course.

closed 2022-05-22 by khaiin