motioneye-project/motioneye #2504: Movies not recorded

Yngve0, 2022-05-22:
I have a raspberry pi 4 8MB running with:
motionEye Version0.43.0
Motion Version4.3.2
OS VersionRaspbian 11 

It is connected to 4x network-cameraes with motiontriggered recording enabled.
Unfortunately, none of them is recording movies. I have tried to change ome of to "Continiuous recording" but still no movies. 

Where could i start digging for this?

MichaIng, 2022-05-22:
To start with, the motionEye logs could help:
```sh
journalctl -u motioneye
```

To rule out issues with bandwidth or so, did you try it with a single netcam only? Also, do those cams require HTTP authentication? We're fixing this currently: #2486

Yngve0, 2022-05-23:
Thanks,
The preview of remote camera is working and I am also able to do interval snapshot from all remote cameras, so it does not looks like authentication-issue
[2:ml2:Camera1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1:Camera1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:Camera2] [WRN] [NET] Above message repeats 7 times

MichaIng, 2022-05-23:
And regarding bandwidth, did you try out with a single camera added only?

Yngve0, 2022-05-23:
> And regarding bandwidth, did you try out with a single camera added only?
I have but still no movies, just realtime view and still images.

May 23 21:23:57 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
May 23 21:23:57 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
May 23 21:23:57 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
May 23 21:24:00 mymotioneyemotion[711]: [1:ml1:myremotecamera] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
May 23 21:24:05 mymotioneyemotion[711]: [1:ml1:myremotecamera] [WRN] [ALL] motion_init: Pass-through processing disabled.
May 23 21:24:10 mymotioneyemotion[711]: [1:ml1:myremotecamera] [WRN] [NET] netcam_next: called with no data in buffer
May 23 21:24:15 mymotioneyemotion[711]: [1:ml1:myremotecamera] [WRN] [NET] Above message repeats 1 times
May 23 21:24:15 mymotioneyemotion[711]: [1:ml1:myremotecamera] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
May 23 21:24:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] Above message repeats 29 times
May 23 21:24:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
May 23 21:24:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
May 23 21:24:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
May 23 21:24:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
May 23 21:30:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] Above message repeats 618 times
May 23 21:30:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
May 23 21:30:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
May 23 21:30:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
May 23 21:30:31 mymotioneyemotion[711]: [1:ml1:myremotecamera] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

MichaIng, 2022-05-24:
> ```
> [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
> ```

Could you try to change the movies format? I think it's H.264/OMX by default, probably this is the issue and H.264 (no OMX). Actually this makes much sense when you have (full) KMS display driver enabled (default since Bullseye), since, if I do not mix it up, OMX/OpenMAX is supported only with legacy display driver or fake KMS. Could be quite the explanation for some other RPi issue reports, and a reason for us to start integrating proper support for modern RPi display driver (full KMS) + changing defaults and clearly marking the old MMAL + OMX support as legacy/deprecated.

hasse-basse, 2022-06-15:
I had the same problem and it was the format I had to change to H.264 (.mp4) The other two H.264 versions, did not work for me. I read somwhere it is connected to a Codec issue.

hoepps2802, 2022-06-30:
@hasse-basse Thank you - it works

MichaIng, 2022-07-01:
We need to change the default to `H.264` (without OMX), at least for non-DSI/`vcgencmd`-detected RPi camera modules.

closed 2026-11-11 by Marijn0