motioneye-project/motioneye #2506: raspberry with MotionEyeos and PIR Sensor

jrm0316, 2022-05-22:
Hi,
I have a raspberry zero with MotionEyeos in the hallway of my building. It works very well, but during the day,
leaves something to be desired, as it cannot always catch whoever is passing in the hallway. The night is different, in view,
that the hallway has a light that automatically turns on when someone is present.
I want it to be more accurate during the day too, so I wanted to connect a pir sensor HC-SR501 (it's a presence sensor
from the raspberry itself), but I plugged it in and at first it didn't work. How do I get MotionEyeos to use
the pir to detect the presence? Has anyone here tested this pir sensor with MotionEye?

Thank you so much.

zagrim, 2022-05-23:
There's no direct support for PIR sensors, but there have been several issues with that topic with some workarounds and/or partial solutions: 
- #31 has some longer discussion and some ideas how to trigger things based on the sensor
- #1138 links to MEOS issue with sample code for reading GPIO as motion trigger (this requires somewhat deeper technical skills to get things running)

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #1138, which tracks PIR sensor support, as zagrim pointed out; the workarounds with a GPIO script are linked there and in #31.

closed 2026-11-10 by JamBalaya56562