motioneye-project/motioneye #2507: Performance

HoneySpotnet, 2022-05-23:
Hey guys i have adual cpu and 187gb of ram and i dedicated it to motioneye i have 47 cameras up and streaming  . Besides the slower stream quality of the cameras them self i get this moment where all cameras go out then the web viewer shows a error then shortly after everything comes back up . on the server side i see that the system did not reboot but did crash the instance of motioneye/docker to where it restarted .. i figured that it might be a cash dump but not sure... What im wondering if  motioneye/docker are using all the resources of the server and im wondering if there is a way to bump up the resources.

MichaIng, 2022-05-25:
> What im wondering if motioneye/docker are using all the resources of the server and im wondering if there is a way to bump up the resources.

With tools like `htop` you should be able to monitor the resource usage of the motionEye container and whether the ones of course system are exhausted.

Is it the new motionEye Python 3 v0.43.0 container or the old Python 2 v0.42.1 one?

HoneySpotnet, 2022-05-28:
i used this https://bkjaya.wordpress.com/2021/02/02/how-to-install-motioneye-cctv-software-on-ubuntu-20-10-groovy/

MichaIng, 2022-05-29:
Ah, the old Docker container. The new one can be pulled here: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

spupuz, 2022-05-31:
stil got problem in streaming whe using the new version.

`    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/4/current/?_=1654012806374&width=0.8&_username=admin&_signature=f529d6118cbd610d0d6326b227dfac1511c3537e (172.17.0.1) 4.57ms
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/3/current/?_=1654012806374&width=0.8&_username=admin&_signature=3f7419ec6cef55914d5270a35e0746b05a66438d (172.17.0.1) 5.04ms
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/6/current/?_=1654012806374&width=0.8&_username=admin&_signature=bd716cac289891308eeda0f76b2e7cd3a1ed51be (172.17.0.1) 5.65ms
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/5/current/?_=1654012806374&width=0.8&_username=admin&_signature=0ae502f5c26b6b0aed6e398a6a9a71b3be06e2d6 (172.17.0.1) 5.63ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /img/top-bar-buttons.svg (172.17.0.1) 11.33ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /img/combo-box-arrow.svg (172.17.0.1) 11.49ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /img/slider-arrow.svg (172.17.0.1) 11.55ms
   ERROR: Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7f0f8a8e51c0>>, <Task finished name='Task-1100' coro=<PictureHandler.current() done, defined at /usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py:93> exception=TypeError('initial_value must be str or None, not bytes')>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 741, in _run_callback
    ret = callback()
  File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
    future.result()
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7f0f8a8e51c0>>, <Task finished name='Task-1125' coro=<PictureHandler.current() done, defined at /usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py:93> exception=TypeError('initial_value must be str or None, not bytes')>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 741, in _run_callback
    ret = callback()
  File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
    future.result()
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7f0f8a8e51c0>>, <Task finished name='Task-1129' coro=<PictureHandler.current() done, defined at /usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py:93> exception=TypeError('initial_value must be str or None, not bytes')>)
[... 7 lines cut ...]
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 741, in _run_callback
    ret = callback()
  File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
    future.result()
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes`

MichaIng, 2022-06-01:
This has been already fixed with #2483. You installed this container mode than 14 days ago, didn't you?

spupuz, 2022-06-01:
> This has been already fixed with #2483. You installed this container mode than 14 days ago, didn't you?

I still got the problem with this container