motioneye-project/motioneye #2511: RTSP stream is stuttering and flaking on motionEye

wecoon, 2022-05-27:
My setup:

Pi Zero with headless Buster streaming RTSP feed from Pi IR Camera
`raspivid -o - -t 0 -n -rot 0 -w 1280 -h 720 -fps 20 -b 3000000 | cvlc -vvv stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h264 --h264-fps=20`

Pi 4B 2GB with headless Bullseye and motionEye webserver running the hub. 

The problem seems to be with motionEye. I have the Pi zero RTSP feed connected as a network camera. Video on motionEye hub is usually just fine, but it does stutter and flake every several seconds. This is not ideal on its own as you may lose some footage, however, this also messes up how motion detection and recording. The problem persists if I close the web client and just leave the hub work on its own, recorded footage is still flaky and stuttering...

On another hand, there are no issues when watching the same stream on VLC (on PC). 
Pi4 is using about 25% CPU, so that should not be an issue. 

On the hub side, there is no special config. I did need to add `rtsp_uses_tcp off` to Extra Motion Options for the camera to start working (so I guess it works on UDP), and movie format is H.264 (H.264/OMX was not able to record)

Also, Pi4 is now connected with eth cable and wifi. Pi Zero only wifi (but the stream is fine on VLC)

I would appreciate any suggestions and help with the issue, but here are a few questions I have:

- **How else I could stream video, so motionEye would be an easier job handling it? I was thinking about TCP, but it seems that motionEye does not support it**
- **Is it possible to stabilize RTSP feed capture?**
- **Maybe I should try Buster on the Hub pi also?**
- **Any lightweight alternatives for motionEye to do motion capture and recording for multiple cameras?**

Thank you all for your help and this nice product. Would love to get it working. 

Some picture how video looks:
[image]
[image]

MichaIng, 2022-05-27:
Which motionEye version do you use? v0.42.1 or v0.43.0 from the `dev` branch? https://github.com/motioneye-project/motioneye/tree/dev#installation

wecoon, 2022-05-27:
motionEye Version 	0.42.1
Motion Version 	4.3.2
OS Version 	Debian 11

MichaIng, 2022-05-27:
Please try it with the Python 3 version of the current `dev` branch. Much has changed, and it would be a waste of time to debug things for the old Python 2 version of motionEye 😉.

wecoon, 2022-05-27:
@MichaIng Thanks. I will try in the following few days hopefully, and let you know

wecoon, 2022-05-29:
I just installed the `dev` branch version, and the performance looks brilliant. Not only the camera picture has no stutters, but the web is loading quicker and feels more responsive. I will continue running the setup, adding more cams maybe and overall testing. Thanks for the help and this brilliant tool.

closed 2022-05-29 by wecoon

MichaIng, 2022-05-29:
Great 🙂.

wecoon, 2022-05-29:
Well, it seems I was too quick to tell the good news. Hub definitely more stable, but I still get these strange video drags/stretching (not really stuttering as before). Noticed that motion jumps above 100%, sometimes to 200% CPU usage when this happens. Maybe because it starts recording. I did test a few different resolutions. Lower was better, I think. 

I use 1280x1024, 20FPS now, but I do not set any bitrate for raspivid. Overall, the hub runs on around 60% CPU, but sometimes jumps above 100%

This was the most extreme, usually, video drags just a bit to trigger motion detection and start recording:
[image]

Here is an example. VLC shows good video, motionEye did drag a bit which triggered recording, but CPU was ok this time:
[image]

Maybe PI4 is just not enough to run motionEye hub?
What logs I should check to find if something is misbehaving? Motion might be having troubles, as sometimes it stays around 60% CPU usage, but sometimes jumps to 100%-200%

MichaIng, 2022-05-29:
What drive and filesystem are you using on the RPi 4 as recordings target? Does it work better with lower CPU usage when you disable recording?

And does the recorded video contain the same artefacts, or only the stream shown in browser?

It would be also a bandwidth limitation since the camera stream is "downloaded" from the RPi Zero and "uploaded" to the browser concurrently.

wecoon, 2022-05-29:
I've just tried with motion and movie disabled, and it looks more stable. It still sometimes lags and gives this square'ish video, but not as bad. After turning on the motion and movie, I noticed, that on movement and recording start, CPU usage jumps above 100%. So maybe, that's just too much for my Pi?

Recorded video does contain the same artefacts, does not matter if you watch on the browser or downloaded file. 

Feed from camera goes through wifi, but it shows ok on VLC, so it is not that. Hub is connected via ethernet cable (but has wifi config too, so I am not 100% sure which connection it is using). Anyway, the issue seems most related to CPU usage as it jumps above 100% when motion triggers the recording.  

Not sure if I understand what you mean with "drive" and "filesystem", but maybe something like this:
Raspberry Pi 4B, 2GB RAM
Raspbian 64bit (Debian 11) lite (headless)
motionEye Version 0.43.0
Motion Version 4.3.2
Root directory for video: /var/lib/motioneye/Camera1 (default)
I am using a cheap Samsung microSD card with a 130MB/s transfer speed. Before I tried another brand with 100MB/s

MichaIng, 2022-05-29:
> Feed from camera goes through wifi, but it shows ok on VLC

But you do not run the VLC player on the RPi 4, do you? However, producing and uploading the camera stream from RPi Zero is hence not the issue.

When not watching the stream via web UI causes the same artefacts on the recording, bandwidth is ruled out as well: Similar to when using VLC player, the stream is uploaded from RPi Zero and downloaded from the client. When watching the stream via motionEye web UI, there are more stages involved: RPi Zero => RPi 4 => browser, i.e. two uploads and two downloads, hence the test.

> I am using a cheap Samsung microSD card

The EVO Plus/Select, I guess? Quality is fine, but the I/O is still limited to ~40 MiB/s on RPi end and as long as it's not the "Endurance" SD card variant, I wouldn't recommend to write video recordings to it. SD cards are simply not made to regular writes and will die fast. If you have any chance, attach a USB flash drive, or an SSD (with dedicated power supply!) for this, which will not only increase I/O but also the SD card lifetime dramatically.

> Anyway, the issue seems most related to CPU usage as it jumps above 100% when motion triggers the recording.

Not necessarily. The max shown CPU usage can be 400% with 4 cores. Recordings are done via FFmpeg, which should be hardware accelerated when the right video format is chosen. Is it **H.264/OMX**?

What I'm wondering is whether OpenMAX is support with RPi Bullseye and KMS video driver. Could you try to either switch to **H.264** without **OMX** in motionEye recording settings, or switch to legacy display driver, e.g. via `raspi-config`?

wecoon, 2022-05-29:
I closed the browser page to test it, CPU usage drops a bit, but the issue persists. 

Yes, it is EVO plus. Thanks for the heads-up, but I am fine to fry the microSD while testing the solution. When I will get stable video, motion detection and recording, I will consider external storage. Can recording to the microSD card be causing the issue?

I actually use H.264 without OMX, because the OMX option does not save recorded videos. 

What do you mean to switch to a legacy display driver? In raspi-config, I can only switch to a legacy camera support. Is this what you have in mind?

I was thinking maybe try running this on Buster instead of the Bullseye. Do you think that might help? However, I don't think 64bit Buster exists. Should it matter if it's 32 or 64bit OS?

Thanks for the support

MichaIng, 2022-05-29:
> When I will get stable video, motion detection and recording, I will consider external storage. Can recording to the microSD card be causing the issue?

Potentially yes, since as mentioned I/O bandwidth to SD card is limited while USB allows higher transfer rates.

> I actually use H.264 without OMX, because the OMX option does not save recorded videos.

Ah, does it throw a related error? That KMS does not support OpenMAX could be actually the reason for this, FFmpeg might simply fail:
```sh
journalctl -u motioneye
```

> I can only switch to a legacy camera support. Is this what you have in mind?

No, there should be a display option to change the driver, with full KMS, fake KMS and legacy as options. This one: https://github.com/RPi-Distro/raspi-config/blob/0319d1c/raspi-config#L1926-L1928
_Ah, seems they removed the fake KMS option, which would be KMS but still with OpenMAX and DispmanX support._

When using the legacy driver, you should be able to use **H.264/OMX** format, which probably reduces CPU usage during recordings.

> I was thinking maybe try running this on Buster instead of the Bullseye. Do you think that might help? However, I don't think 64bit Buster exists. Should it matter if it's 32 or 64bit OS?

It shouldn't make a difference but only provides you with old software/package versions. The main difference is that Raspberry Pi OS Buster images by default come with fake KMS enabled for RPi 4, hence OpenMAX support, Bullseye images use the full KMS driver by default instead. But since you can switch it via `raspi-config` (or manually in `/boot/config.txt`), there is no need to switch to an old image version for this.

32-bit vs 64-bit shouldn't have any significant impact on performance. The main benefit of the 64-bit image is that it doesn't use the Raspbian repository, but the original Debian repo, which provides the dedicated security repo for faster security updates, as well as backports, and it is more stable (less inconsistencies during repo syncs or faulty/incomplete syncs), from what I can say. Performance-wise it is a trade-off between slightly higher RAM usage with faster 64-bit arithmetic, but both should have a recognisable impact for motionEye.

wecoon, 2022-05-29:
Thanks @MichaIng I will go back to the bullseye because buster is performing even worse actually. Then I will set up everything again and try to check for errors, also I will try to change display options, but I did not find them before under `raspi-config`. I will google, but if you could elaborate a bit on how to enable legacy drivers, it would be really helpful. 

I will edit with the results hopefully later today

MichaIng, 2022-05-29:
> I will try to change display options, but I did not find them before

Ah I see now it is hidden on RPi 4, for whatever reason 🤔. Do this to switch to legacy display driver:
```sh
sed -Ei 's/^dtoverlay=vc4-(f?)kms-v3d/#dtoverlay=vc4-\1kms-v3d/' /boot/config.txt
reboot
```

wecoon, 2022-05-29:
`May 29 21:14:03 raspberrypi systemd[1]: Started motionEye Server.
May 29 21:14:05 raspberrypi meyectl[445]: configure_logging cmd motioneye: False
May 29 21:14:05 raspberrypi meyectl[445]: configure logging to file: None
May 29 21:14:05 raspberrypi meyectl[445]:     INFO: hello! this is motionEye server 0.43.0
May 29 21:14:06 raspberrypi motion[497]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 29 21:14:06 raspberrypi motion[497]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 29 21:14:06 raspberrypi motion[497]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 29 21:14:06 raspberrypi motion[497]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 29 21:14:06 raspberrypi motion[497]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 29 21:14:06 raspberrypi motion[497]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 29 21:14:06 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Network is unreachable
May 29 21:14:06 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
May 29 21:14:06 raspberrypi motion[497]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
May 29 21:14:06 raspberrypi motion[497]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
May 29 21:14:08 raspberrypi meyectl[445]:     INFO: cleanup started
May 29 21:14:08 raspberrypi meyectl[445]:     INFO: wsswitch started
May 29 21:14:08 raspberrypi meyectl[445]:     INFO: tasks started
May 29 21:14:08 raspberrypi meyectl[445]:     INFO: mjpg customer garbage collector has started
May 29 21:14:08 raspberrypi meyectl[445]:     INFO: server started
May 29 21:14:10 raspberrypi motion[497]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
May 29 21:14:10 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Network is unreachable
May 29 21:14:10 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
May 29 21:14:10 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Network is unreachable
May 29 21:14:10 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
May 29 21:14:10 raspberrypi motion[497]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
May 29 21:14:10 raspberrypi motion[497]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
May 29 21:14:30 raspberrypi motion[497]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
May 29 21:14:39 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
May 29 21:14:39 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
May 29 21:14:39 raspberrypi motion[497]: [1:ml1:Camera1] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
May 29 21:15:06 raspberrypi motion[497]: [2:nc2:Camera1] [ERR] [NET] netcam_rtsp_handler_reconnect: Normal resolution: Reconnecting with camera....
May 29 21:15:14 raspberrypi motion[806]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 29 21:15:14 raspberrypi motion[806]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 29 21:15:14 raspberrypi motion[806]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 29 21:15:14 raspberrypi motion[806]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 29 21:15:14 raspberrypi motion[806]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 29 21:15:14 raspberrypi motion[806]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 29 21:15:48 raspberrypi motion[806]: [2:nc2:Camera1] [ERR] [NET] netcam_rtsp_handler_reconnect: Normal resolution: Reconnecting with camera....
May 29 21:16:14 raspberrypi motion[806]: [1:ml1:Camera1] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
May 29 21:16:14 raspberrypi motion[806]: [1:ml1:Camera1] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
May 29 21:16:14 raspberrypi motion[806]: [1:ml1:Camera1] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
May 29 21:16:21 raspberrypi motion[806]: [2:nc2:Camera1] [ERR] [NET] netcam_rtsp_handler_reconnect: Normal resolution: Reconnecting with camera....
`
I've tried the command you have sent, but it does not seem to help in this case. I checked config.txt and it had `dtoverlay=vc4-fkms-v3d` commented out. Uncommenting it did no difference. 

In the log, there are a few ffmpeg errors which seem to disappear if I change to H.264 without OMX

However, this one persists, maybe an issue handling the connection with the camera?
[2:nc2:Camera1] [ERR] [NET] netcam_rtsp_handler_reconnect: Normal resolution: Reconnecting with camera....

MichaIng, 2022-05-29:
Hmm, I would have expected the `Encoder not found` to disappear as well when having `dtoverlay=vc4-fkms-v3d`, actually even when having it uncommented. The one which doesn't support OMX, if I'm not mistaken, is the full KMS driver: `dtoverlay=vc4-kms-v3d` (without the `f` for "fake").

Could you try the other way round, now that you use `H.264` (without OMX), adding `dtoverlay=vc4-kms-v3d` to use the full KMS driver? On Bullseye this should actually serve hardware acceleration as well.

The `Reconnecting with camera` "error" actually doesn't look like an error, more like an info message which has a wrong/unexpected log level. Makes sense that it reconnects as a result of changing settings which implies a motion restart.

wecoon, 2022-05-30:
Sadly, tried all different ways, without any success.
H.264 with OMX always gives the error, despite if I set `dtoverlay=vc4-fkms-v3d` or `dtoverlay=vc4-kms-v3d` or comment both out.

H.264 without OMX, with `dtoverlay=vc4-kms-v3d` gives no errors, but the video feed has the same error as in the screenshots above.

Actually, it does not matter what I set, the issue is still there. Even if I turn off motion detection and recording, the issue is still there, but just not that noticeable.

I think the issue is with how motionEye is handling RTSP feed.