motioneye-project/motioneye #2514: Python 3.10 seems to break this.

JaseNZC, 2022-06-03:
Anybody have this working on python 3.10 system ??

To me its seems upgrading from 3.9 to 3.10 breaks it.

I don't know how its breaking it maybe its just me.

Won't work on latest Unbutu

JaseNZC, 2022-06-03:
Thanks at least I know now it was not just my setup.

MichaIng, 2022-06-03:
I didn't recognise any issues on Debian Bookworm (Python 3.10) so far. On Debian and Ubuntu `python` is always Python 2 and `python3` hence needs to be used.

In general, for troubleshooting, we need some info, logs (`journalctl -u motioneye`) or at least a closer description what fails where, more than just "breaks it" 😉.

JaseNZC, 2022-06-04:
> I didn't recognise any issues on Debian Bookworm (Python 3.10) so far. On Debian and Ubuntu `python` is always Python 2 and `python3` hence needs to be used.
>
> In general, for troubleshooting, we [...]

Yeah sorry my description was about as good as a plank of wood.
I have resolved the Ubuntu issue that was user error and nothing to do with python is seems.

I do have a problem with Kali linux though.

A clean install of 2022.2 ends with an error while trying to install.

I don't know why its throwing a recursion error , its running python 3.10.4

[error.txt](https://github.com/motioneye-project/motioneye/files/8836646/error.txt)

zagrim, 2022-06-04:
[motioneye-python3.10.4-installfail.log](https://github.com/motioneye-project/motioneye/files/8837129/motioneye-python3.10.4-installfail.log)
Attached full log on doing `python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'` (without sudo though since I don't want system wide install on this box, but the error doesn't seem to relate to permissions etc). Doing the same by substituting `python3.9` with `python3` works just fine.

Here's the error
`     Traceback (most recent call last):
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 94, in _load_sysconfig_schemes
          for scheme in sysconfig.get_scheme_names()
        File "/usr/lib/python3.10/sysconfig.py", line 583, in get_scheme_names
          return tuple(sorted(_INSTALL_SCHEMES))
      RecursionError: maximum recursion depth exceeded while calling a Python object
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 261, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 230, in build_wheel
          return self._build_with_temp_dir(['bdist_wheel'], '.whl',
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 215, in _build_with_temp_dir
          self.run_setup()
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 158, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 1, in <module>
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 148, in setup
          return run_commands(dist)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 163, in run_commands
          dist.run_commands()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 967, in run_commands
          self.run_command(cmd)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 335, in run
          self.run_command('install')
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 985, in run_command
          cmd_obj.ensure_finalized()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 107, in ensure_finalized
          self.finalize_options()
        File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 45, in finalize_options
          orig.install.finalize_options(self)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 320, in finalize_options
          self.finalize_unix()
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 59, in finalize_unix
          super().finalize_unix()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 488, in finalize_unix
          self.select_scheme("posix_prefix")
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 55, in select_scheme
          super().select_scheme(name)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 522, in select_scheme
          scheme = _load_schemes()[name]
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 137, in wrapped_load_schemes
          _inject_headers(name, scheme)
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 125, in _inject_headers
          scheme['headers'] = orig_install._load_schemes()['posix_prefix']['headers']
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 137, in wrapped_load_schemes
          _inject_headers(name, scheme)
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 125, in _inject_headers
          scheme['headers'] = orig_install._load_schemes()['posix_prefix']['headers']
        File "/usr/lib/python3.10/_distutils_system_mod.py", line 137, in wrapped_load_schemes
          _inject_headers(name, scheme)
        [...and so it goes on until max call recursion is met...]
`

zagrim, 2022-06-04:
Now, in my case, this is not by far a clean install of Debian testing ("bookworm/sid" says `/etc/debian_version`) since it has been upgraded over several releases, and `pip` is from Debian packages:
```
$ pip --version
pip 22.1.1 from /usr/lib/python3/dist-packages/pip (python 3.10)
```
So that might be the issue for me, somehow. But it's not an issue really for me as far as there's something else, too, that relies on Python 3.9 on this box, which I guess will be a long time.

JaseNZC, 2022-06-04:
Doing a clean install of bookworm it install's just fine except for permissions for the log file which I guess should be addressed in the install routine as it won't start because of it and an unexperienced user may not know how to resolve it.
Wont install on Kali though with the same error as @zagrim.
I thought kali was based off of debian so don't know why it does not like it.
This is the permission error from a fresh install of either Debian or Ubuntu from both their respective latest versions.

[motion.txt](https://github.com/motioneye-project/motioneye/files/8837557/motion.txt)

MichaIng, 2022-06-04:
This is a bug in the `motion` package, not related to motionEye and not relevant, since motionEye starts its own `motion` processes(s). The global one shouldn't be running anyway:
```sh
systemctl disable --now motion
```

JaseNZC, 2022-06-04:
> This is a bug in the `motion` package ([Motion-Project/motion-packaging#15](https://github.com/Motion-Project/motion-packaging/issues/15)), not related to motionEye and not relevant, since motionEye starts its own `motion` processes(s). The global one shouldn't be running anyway:
> 
> ```shell
> systemctl disable --now motion
> ```

Excellent thanks for that :-)

There is just the other fault now to resolve and I would be a happy man.

JamBalaya56562, 2026-11-10:
Closing, as current motionEye is tested on Python 3.9 to 3.15 and installs into a venv as the README describes, which avoids the distribution's own pip and setuptools that failed here in 2022. The log permission error was the separate `motion` service, which should stay disabled, as MichaIng explained.

closed 2026-11-10 by JamBalaya56562