motioneye-project/motioneye #2525: Real time movie

Avessalom608, 2022-06-10:
How to open a movie in real time before it is fully recorded?
I don't want to wait for the full movie

Rassberrpy pi 3b+

motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Raspbian 11
[image]

MichaIng, 2022-06-13:
It depends on the file format whether this is possible at all. Watching the live stream in web UI is not an option?

Avessalom608, 2022-06-14:
@MichaIng The fact is that sometimes you need to watch the recording, but the file is 2 hours long. I can certainly reduce the recording time, but it's inconvenient

JamBalaya56562, 2026-11-13:
Whether a movie can be opened while Motion is still writing it depends on the format, as @MichaIng said. For long recordings, a lower *Maximum Movie Length* splits them into shorter movies that can be watched sooner, and the live stream shows what is happening now.

closed 2026-11-13 by JamBalaya56562