motioneye-project/motioneye #2526: E: Unable to correct problems, you have held broken packages.

superidol5, 2022-06-11:
I installed the dev motioneye by the dev wiki and when i got to the `sudo motioneye_init` I've got that error:
[image]

os: Volumio
device: Raspberry pi

coder0107git, 2022-06-11:
Run `sudo apt install libavcodec58 libavdevice58 libavfilter7 libavformat58 libswresample3` and then rerun `sudo motioneye_init`.

superidol5, 2022-06-13:
Some basic Git commands are:
```
git status
git add
git commit
```

superidol5, 2022-06-14:
> ```shell
> apt policy libavcodec58
> dpkg -l | grep libavcodec
> ```

```
volumio@sus:~$ apt policy libavcodec58
dpkg -l | grep libavcodeclibavcodec58:
  Installed: (none)
  Candidate: 7:4.1.9-0+deb10u1+rpt1
  Version table:
     7:4.1.9-0+deb10u1+rpt1 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
     7:4.1.9-0+deb10u1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
volumio@sus:~$ dpkg -l | grep libavcodec
volumio@sus:~$
```

JamBalaya56562, 2026-11-10:
Closing, as this was the 2022 install script running into Volumio's own ffmpeg libraries on Buster. The README now installs motionEye into a venv with the distribution's `motion` package; please open a new issue if that still fails on a current system.

closed 2026-11-10 by JamBalaya56562