motioneye-project/motioneye #2530: Adding Mjpeg camera as network one: Corrupted video

foxtreat, 2022-06-12:
Greetings
im using Motioneye Installed on a Pi4
but im having a severe issue:
adding my Ip camera ( a Dlink DCS-932L) as a Simple Jpeg camera works fine
but adding it as a Network camera (and having it detected as a Mjpeg) leads to a severe graphical corruption of the Online Video Stream
tested with all links from the camera itself(specific to this model), the corruption does not stay when accessing the camera directly
appear like a burn-in : the camera shows a ghost image of what it originally saw, and gets darker/more baked in within seconds
the camera view is utterly unusable in this state
Motion is up to date, all packages from the install are upgraded 

here is the camera view( added 5 times using different access links, all are working in the browser, the first 2 views are from the "add simple Jpeg camera" option, all the black ones are the "add network camera" option

[image]

this is the view 3 seconds after covering and uncovering the lense
[image]

MichaIng, 2022-06-13:
Which motionEye version are you using, v0.42.1 or v0.43.0 dev version already?

JamBalaya56562, 2026-11-10:
Closing, as this got no answer to which version it was. The same camera works as a Simple MJPEG Camera, which is the type to use for it when Motion's decoding of its stream breaks up. Please open a new issue with `motion.log` if a Network Camera still shows that with a current release.

closed 2026-11-10 by JamBalaya56562