motioneye-project/motioneye #2531: Dropbox issue Authorization Key

hasse-basse, 2022-06-13:
The Dropbox Authorization Key doesn't last long time, it seem like it lasts 24 hours or so. I have had the same issue with Virtualmin when doing backups to Dropbox and was told by the Dropbox staff the they have changed how things work. Please see their answer about Virtualmin. With this information Virtualmin could solve the problem, and I guess you can as well:

> You’ll need to contact Virtualmin. Dropbox retired long-lived access tokens, so Virtualmin will likely need to update their app to use refresh tokens instead. See this link [Migrating App Permissions and Access Tokens - Dropbox 2](https://dropbox.tech/developers/migrating-app-permissions-and-access-tokens)

**So you will have to update the Motioneye to use refresh tokens insted of  long-lived access tokens**

I'm very keen of this being arranged as it is impotant to me to backup to Dropbox.

byWulf, 2022-07-09:
Has been fixed in https://github.com/motioneye-project/motioneye/pull/2396 already, but sadly not yet in an official release.

hasse-basse, 2022-07-10:
Ok, thanks!

closed 2022-07-10 by hasse-basse