motioneye-project/motioneye #2533: not supported or even wiped out when adding manually:: output_pictures center

ufehembach, 2022-06-15:
Hi, the feature "output_pictures center" is not supported and manual changes to the conffiles to add this are removed when changing the config.

MichaIng, 2022-06-19:
Recent `motion` uses `picture_output`, `output_pictures` was the name until motion v4.1.

`center` however is not valid value: https://motion-project.github.io/motion_config.html#picture_output

The feature is basically controlled within the still images section in motionEye GUI. It is expected that the individual camera motion configs are rewritten when you change and save settings in motionEye.

ufehembach, 2022-06-19:
Hi, thanks for the quick response.
I'm using motion 4.1 and center is still in the sample config file.
There is no more recent version of motion available on ubuntu mate on a raspi4. Compile failed horribly.
may is ask, what the feature/field is called in the still images section of motionEye to select the image type? I have

Image File Name 
Image Quality
Capture Mode
Preserve Picture
Enable manual snapshot

also I played with the feature "extra motion option" in "video device". This made motioneye stop and losing all the config.

JamBalaya56562, 2026-11-10:
Closing, as MichaIng answered this: since Motion 4.1 the option is `picture_output`, which has no `center` value, and in motionEye it is set by Still Images > Capture Mode; motionEye rewrites Motion's config files from its settings, so edits made to them by hand are replaced.

closed 2026-11-10 by JamBalaya56562