motioneye-project/motioneye #2535: Main Dashboard has garble video and it also shows on the snapshot URL

kd7hhw, 2022-06-16:
First, I want to thank you for the application.  It is great!  If I could get a Pi3 or Pi4 for a reasonable price, I might be using the motioneyeos.

Yesterday, I had everything working.  I could look at the dashboard and see the cameras just fine.  It worked this morning too.  This after noon, I made a change to the motion detection or 1 camera (I have 2 total).  I moved the slider from 20% to 10%.  Since then, the dashboard and snapshot URL show a garbled picture for both cameras.

I went as far as deleting the OS and starting fresh, which didn't help.

The stream URL looks great.  If I go to the links for the camera (one is rtsp and the other http), they look great.  

I have the following:
motionEye Version | 0.42.1
Motion Version | 4.4.0+git20220612-1cc4018
OS Version | Ubuntu 22.04

Attached is an example garbled picture.  Note, when on the dash board, the garble picture is always changing to a different garbled look.

I can do anything to this system (blow it away and start again) if we can get it resolved.

Thanks!
Matt

[image]

MichaIng, 2022-06-19:
Note that motionEye does not fully support motion v4.4 yet. WIP: https://github.com/motioneye-project/motioneye/pull/2462

Also, please try the `dev` branch, which contains a lot of fixes: https://github.com/motioneye-project/motioneye/tree/dev#installation

kd7hhw, 2022-06-22:
I went back to motion v4.3 and all is well for now.  Hoping the ubuntu updates won't update it to a version that doesn't work.  Thanks for the response and resolution!

closed 2022-06-22 by kd7hhw