motioneye-project/motioneye #2537: Motion detection

kapabel, 2022-06-19:
Gday!
I have a very hard time to set up motion detection. I tried tinkering around a long while but can not make it work reliably. Either the wind in the bushes fire the motion detection too early or the person doesn’t get recognised.

Here is a screenshot of my camera. With two kids in the frame to have an idea of my setup.

[image]

What settings would you choose when I want to register the motion of the people walking in? just roughly, i will then fine tune of course. I want to register 5 frames before the motion, then the entire motion and then 5 frames after.

I also noticed that for some reason sometimes the video will not be recorded for the entire motion but cuts off earlier.

here are my settings I have at the moment:

Frame Change Threshold: 7%
Maximum Change Threshold: 1pixels	
Auto Threshold Tuning: OFF	
Auto Noise Detectionm ON	
Light Switch Detection: 0%
Despeckle Filterm ON	
Motion Gapm 30 seconds	
Captured Before: 5 frames	
Captured After: 5 frames	
Minimum Motion Frames: 2 frames	
Mask: OFF	
Show Frame Changes: OFF	
Create Debug Media Files: OFF

thanks a million!

zagrim, 2022-06-19:
I'd definitely recommend setting up the motion detection mask to mask of those areas with foliage which are out of the path of people walking in (right side of the image mostly, the parts where foliage already prevents the camera from seeing anything else), that should reduce the ratio of false detections quite much. After setting up the mask you probably need to re-tune the Frame Change Threshold somewhat since it is no longer necessary to compensate for the non-interesting motion.

This seems odd to me:

> Maximum Change Threshold: 1pixels

The tooltip (and the [docs](https://motion-project.github.io/motion_config.html#threshold_maximum)) says 0 disables the setting, but having 1 would mean that if there's anything more that one pixel change between the frames, it is not considered as motion, which surely wouldn't work as you hope it to work.

> I also noticed that for some reason sometimes the video will not be recorded for the entire motion but cuts off earlier.

You might want to check that the "Maximum Movie Length" under "Movies" section is not too short, and maybe you also could just increase "Captured After" to compensate the possibility of having less detected motion in the end of the event when the person is still visible. Anyway, all of those settings where the unit is "frames" depend on the framerate of the video device, and what you generally want in "before"/"after" is just something that equals from zero to a couple of seconds worth of video (number of seconds times the camera FPS), to get not too much "empty" video but also to not miss anything interesting in the start or the end of the video. 

A bit depending also on the light conditions having "Light Switch Detection" enabled might help to avoid recording sudden changes of light on a semi-cloudy day, but that's more like fine tuning.

As for the general limitations of the detection algorithm of Motion [I wrote some points a while back](https://github.com/motioneye-project/motioneye/issues/2455#issuecomment-1115725208). Seeing your scene I don't think your setup is nearly as difficult as the one documented in that issue, though.

kapabel, 2022-06-19:
Oh this is gold! Thank you so much for this valuable information and links. I understood the max change threshold wrong and changed it to 0. I will mask out the leaves as much as practical. Thanks so much, very helpful.

closed 2022-06-19 by kapabel