motioneye-project/motioneye #2540: Running motioneye docker as non-root user, can't access /var/run

towerofpower256, 2022-06-22:
If I installed the motioneye docker and ran as root, no issues there.

However, it I use the motioneye docker image as another user that is not root (e.g. `--user 1003:1003`), I started getting errors that are visible in the log saying that motioneye was unable to create the .pid file in the /var/run directory. Motioneye isn't able to start either, and I can't get to the web interface.

The log:
```
CRITICAL: pid directory "/etc/motioneye/run" does not exist or is not writable
    INFO: hello! this is motionEye server 0.42
CRITICAL: pid directory "/etc/motioneye/run" does not exist or is not writable
    INFO: hello! this is motionEye server 0.42
CRITICAL: pid directory "/etc/motioneye/run" does not exist or is not writable
    INFO: hello! this is motionEye server 0.42
```

Here's my docker run command, including `--user` and RUN_UID and RUN_GID arguments:

```sh
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /srv/motioneye/etc:/etc/motioneye \
    -v /srv/motioneye/lib:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    --user 1003:1006 \
    -e "RUN_UID=1003" \
    -e "RUN_GID=1006" \
    ccrisan/motioneye:master-amd64
```

I was able to get around this by updating motioneye.conf to store the pid file in another directory that motioneye should have access to. This fix appears to be working so far.

```sh
# path to the directory where pid files go (must be writable by motionEye)
#run_path /var/run
run_path /etc/motioneye/run # Fix for no access to /var/run within container if non-root
```

I suspect this is due to the Docker image that the motioneye docker image is utilizing, `debian:buster-slim`. 
https://github.com/debuerreotype/docker-debian-artifacts/blob/337f494fae12a1db13a003cea38e74f43d312ee6/buster/slim/rootfs.tar.xz

If I have a look at the file structure that it unpacks:
* `/run` exists, with permissions of 755 (writable by root, read-only to everything else).
* `/var/run` exists appears to be a link to `/run`.

I'm wondering if there's a way to include something in the motioneye docker image to link `/var/run` to `/run`, and allow everyone to both read and write to it? Or to automatically update the motioneye.conf in the docker image to save the pid file in another directory like `/etc/motioneye`?

towerofpower256, 2022-06-22:
Possibly related to:
#2407 
#2388

towerofpower256, 2022-06-22:
This article here suggests that it may be better to create a directory within `run` to hold app-specific pid files.
https://serverfault.com/questions/159334/what-permissions-are-needed-to-write-a-pid-file-in-var-run

E.g. a new directory `/var/run/motioneye`, and update the motioneye.conf to store the pid files in there, instead of just `/var/run`.

Ashcora, 2022-11-08:
I am facing the same issues.
Via Portainer the log is showing the following after reboot (as assumed with regard to the article):

 INFO: hello! this is motionEye server 0.42.1
CRITICAL: pid directory "/var/run/motioneye" does not exist or is not writable

So I wrote a startup script as linked in the article:
(thanks @towerofpower256 )

mkdir /var/run/motioneye
chown motioneye:motioneye /var/run/motioneye

My questions:
- Will/can this task (adding the folders and permissions after reboots) be part of the docker container in future?
- How "dangerous" is it to run the container as root? (without adding the UID and GID to docker run and modifying the /var/run)

PhatHub, 2023-02-20:
For those who haven't figured out the systemd-onic (sorry didn't know how to say "systemd" in the same way of "pythonic") way of doing it, I just found out... utilize [`tmpfiles.d`](https://www.freedesktop.org/software/systemd/man/tmpfiles.d.html) 

It took me a while because the first time I tried to google for the answer, it took me weeks, and I couldn't find anything besides "it'll be complex."

A few weeks later I found [this answer within minutes](https://askubuntu.com/questions/303120/how-folders-created-in-var-run-on-each-reboot), and it all clicked.

I did this on Ubuntu Server, so YMMV if you're on something else, like RedHat. `tmpfiles.d` is an installable package, so if it doesn't work, make sure it's installed but I doubt it'll be missing since basically everyone is running `systemd`.

1. Create a motioneye file `sudo touch /usr/lib/tmpfiles.d/motioneye.conf`
2. Edit the file to create a temp file `sudo vim /usr/lib/tmpfiles.d/motioneye.com` (or use your preferred editor instead of `vim`)
3. ...and add a single line, `d /var/run/motioneye 0744 motion motion` (I put tabs instead of spaces to kind of line up columns in the future)
4. Exit your editor, stop motioneye `systemctl stop motioneye` to force it to delete the previous pid directory.
5. Restart it and see if it works. `systemctl start motioneye`

Hopefully this helps.

JamBalaya56562, 2026-10-30:
This works on dev now: #3478 removed the PID file and #3484 supports `docker run --user`. I built the image from current dev and ran it with `--user 1003:1003` and volumes owned by that user, as in [docker/README.md](https://github.com/motioneye-project/motioneye/blob/dev/docker/README.md): the server answers on 8765 and runs as uid 1003, with no PID error. This will be in the next release. Closing; please reopen if it still fails there.

closed 2026-10-30 by JamBalaya56562