motioneye-project/motioneye #2541: Changing movies recording mode from "Motion Triggered" to "Continuous Recording" should not set Still Images capture mode to "All Frames"

JRogaishio, 2022-06-22:
**The Issue**
When changing movies from "Motion Triggered" to "Continuous Recording" and you have Still Images enabled, the UI will automatically change Still Images capture mode to "All Frames"

Still Images capture mode of "All Frames" should be highly discouraged. Even on relatively powerful hardware (compared to a rPi) the system struggles to show a stable video stream. After <1 hr the video preview (of **only** the camera with all frames enabled) in the web dashboard will become unstable / show garbage & compression issues, and then finally outright freeze. The freeze is not recovered and will stay frozen for days until either a setting is changed or the motioneye service is restarted.

**Changes Requested**
I'd recommend when updating from motion triggered to continuous movies the still images capture mode should be set to manual instead. Side issue is the UI doesn't allow the combination of continuous recording movies & manual still images due to the UI changing values around. (Setting still images to manual auto-sets movies to motion triggered, setting movies to continuous will set images to all frames)

**To Reproduce**
1. Enable movies & still images
2. Set Movies to "Motion Triggered"
3. Set Still Images to "Motion Triggered (One Picture)
4. Go back to Movies and set recording mode to "Continuous"
5. Still images capture mode has now auto-changed to "All Frames"
6. After <1 hr the video preview on the web dashboard will become unstable and outright freeze. The freeze is not recovered and will stay frozen for days until either a setting is changed or the motioneye service is restarted

**System Specs:**
OS: Ubuntu 20.04.4 LTS
CPU: Intel(R) Xeon(R) CPU E31240 @ 3.30GHz (4 core 8 thread)
RAM: 16GB
Storage: 16TB (2x 8TB 5400rpm drives in RAID0 because capacity > fault tolerance for me)
Cameras: 5 total RTSP streams from Various Amcrest PoE cameras.

zagrim, 2022-06-23:
Thanks for such a good report! I think you are not the first to request something like this (and I think it would be nice if those two things wouldn't be dependent on each other if technically possible), but instead of looking for old issues for any reasons why it is like it is, I instead went for the code and Motion docs.

Video recording mode "continuous" is implemented by setting `emulate_motion on` (Motion after all is a motion capture application, so the only way to have it record continuously is to have it pretend there is motion). However, as far as I see there shouldn't be any reason why to stick with capture mode of "all frames", it should be possible to have Motion not to capture any pictures (set `picture_output off` but leave `emulate_motion` untouched contrary to how it currently is done) and let the user do manual snapshots. I didn't really dig into the manual snapshot code but that functionality seems to not be related to Motion, so the dependence should not be there.

JRogaishio, 2022-06-23:
> Thanks for such a good report!

Thank you! I was trying to format my issue in a way that's easy to read. Surprisingly motioneye-project doesn't have issue templates setup. (fyi Repo settings -> General -> Features Group -> Issues -> Set up templates)

> Motion after all is a motion capture application, so the only way to have it record continuously is to have it pretend there is motion

It bet it was easier to implement this way but I feel like "Continuous Recording" should just bypass motion entirely and save the video stream directly.... which I'm certain is A LOT of work / changes. However decoupling motion from "Continuous Recording" might also make it possible to have Motion Triggered Still Images which would be handy for skimming large recordings to find "areas of interest". 100% should be its own issue if it's worth diving into.

Unrelated but having `emulate_motion on` 100% of the time might be an issue for integrations (like HomeAssistant) and triggers

> it should be possible to have Motion not to capture any pictures (set picture_output off but leave emulate_motion untouched contrary to how it currently is done) and let the user do manual snapshots

As long as Still images aren't auto-changed to "All Frames" when you change other settings I'll take it. It'd be nice for motion triggered snapshots on continuous recording but that's a stretch

Unrelated but I plan on diving into the web side of motioneye to add additional functionality under "General Settings". I **really** want to see cpu / memory usage on the web ui instead of having to ssh into the server.

rratch, 2025-03-04:
Has this issue been resolved, or a workaround discovered? It's killing me on drivespace. 

If a workaround was described above as the emulate_motion on setting, it's not apparent in the GUI to me, and unclear what impact changing it to "off" will have on video recording.

zagrim, 2025-03-05:
What I wrote above about `emulate_motion` was not describing a workaround, but how "continuous recording" is currently implemented. I don't think there is any workaround, since the logic in MotionEye conflicts with any attempts to try to alter relevant config options manually. Someone should just find the time and energy to try out a change to how the Motion configs are changed when recording mode is changed in the UI.