motioneye-project/motioneye #2544: Adding Annke I51DM to motioneye

9no2, 2022-06-27:
Hello,
i have the I51DM security camera (C500) from Annke. I checked the netcam_url in vlc and found

rtsp://user:password@192.168.0.6:554/Streaming/Channels/101
rtsp://user:password@192.168.0.6:554/user=user_password=password_channel=1_stream=1.sdp
rtsp://user:password@192.168.0.6:554/1

rtsp://192.168.0.6:554/Streaming/Channels/101             #(here i have to enter the user and password manually in vlc)
rtsp://192.168.0.6:554/channel=1_stream=1.sdp            #(here i have to enter the user and password manually in vlc)
rtsp://192.168.0.5:554/1                                                 #(here i have to enter the user and password manually in vlc)

all these options are working in vlc 

------

**My Problem is :** In Motioneye no rtsp address is working. 

[
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Raspbian 10
]

_> pi@raspberrypi:/etc/motioneye $ motion -c /etc/motioneye/camera-1.conf
> [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/camera-1.conf
> [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "rtsp_transport"
> [0:motion] [NTC] [ALL] motion_startup: Using default log level (NTC) (6)
> [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
> [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
> [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
> [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
> [0:motion] [NTC] [STR] webu_start_strm: Starting all camera streams on port 10001
> [0:motion] [NTC] [STR] webu_start_strm: Unable to start stream for camera 0
> [0:motion] [NTC] [ENC] ffmpeg_global_init: ffmpeg libavcodec version 58.35.100 libavformat version 58.20.100
> [0:motion] [NTC] [ALL] translate_init: Language: English
> [0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 0 is from /etc/motioneye/camera-1.conf
> [0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 0 Camera Name: Camera1 Service: rtsp:
> [0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 27046
> [1:ml1:CamCam] [NTC] [ALL] motion_init: Camera 0 started: motion detection Enabled
> [1:ml1:CamCam] [NTC] [VID] vid_start: Opening Netcam RTSP
> **[1:ml1:CamCam] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Invalid argument**
> [1:ml1:CamCam] [NTC] [NET] netcam_rtsp_cleanup: Normal resolution: Shut down complete.
> [1:ml1:CamCam] [ERR] [VID] vid_start: Netcam RTSP failed to open
> [1:ml1:CamCam] [WRN] [ALL] motion_init: Could not fetch initial image from camera
> [1:ml1:CamCam] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
> [1:ml1:CamCam] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
> [1:ml1:CamCam] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 21 items
> [1:ml1:CamCam] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
> [1:ml1:CamCam] [NTC] [VID] vid_start: Opening Netcam RTSP
> [1:ml1:CamCam] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Invalid argument
> [1:ml1:CamCam] [NTC] [NET] netcam_rtsp_cleanup: Normal resolution: Shut down complete.
> [1:ml1:CamCam] [ERR] [VID] vid_start: Netcam RTSP failed to open
> [1:ml1:CamCam] [NTC] [ALL] motion_loop: Thread exiting
> [0:motion] [NTC] [ALL] main: Motion thread 1 restart
> [0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 0 is from /etc/motioneye/camera-1.conf_

Can someone help me with this issue and know what to do ?

_> /etc/motioneye/camera-1.conf
> threshold_maximum 0
> stream_quality 85
> threshold 1999
> noise_level 31
> smart_mask_speed 0
> pre_capture 1
> movie_codec mp4:h264_omx
> noise_tune on
> stream_maxrate 5
> netcam_url rtsp://192.168.0.6:554/channel=1_stream=1.sdp
> netcam_userpass user:password
> rtsp_transport tcp
> stream_localhost off
> text_changes off
> movie_filename %Y-%m-%d/%H-%M-%S
> movie_max_time 0
> lightswitch_percent 0
> movie_passthrough off
> event_gap 30
> auto_brightness off
> stream_port 8082
> rotate 0
> stream_auth_method 0
> threshold_tune off
> framerate 2
> emulate_motion off
> movie_output off
> picture_quality 85
> snapshot_filename
> despeckle_filter
> snapshot_interval 0
> minimum_motion_frames 20
> stream_motion off
> target_dir /var/lib/motioneye/Camera1
> movie_output_motion off
> post_capture 1
> stream_authentication user:
> on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t$
> on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
> text_left Camera1
> picture_output_motion off
> picture_filename
> text_scale 1
> locate_motion_style redbox
> locate_motion_mode off
> netcam_keepalive off
> movie_quality 75
> picture_output off
> on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
> text_right %Y-%m-%d\n%T
> on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
> camera_name Camera1
> netcam_tolerant_check on
> height 480
> mask_file_

9no2, 2022-07-02:
or is the cam not compatible ?

MichaIng, 2022-07-03:
Could you try the new Python 3 based motionEye version from `dev` branch: https://github.com/motioneye-project/motioneye/tree/dev#installation

9no2, 2022-07-04:
Hello Michalng, thanks for your help. I installed it an tried it, but it isn't working. Do you have another idea? Here is the motion logfile:

motionEye Version | 0.43.0
Motion Version | 4.3.2
OS Version | Raspbian 10

python3  (3.7.3-1)

/var/log/motion.log
> [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Invalid argument
[1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Invalid argument
[1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
[1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Invalid argument
[1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera

Pyanne, 2022-10-29:
From the user manual, the rstp should look like this:

rtsp://user:password@192.168.0.6:554/H264/ch1/main/av_stream

JamBalaya56562, 2026-11-11:
Closing, as this got no reply to Pyanne's suggestion: the camera's manual gives `rtsp://user:password@<ip>:554/H264/ch1/main/av_stream` as its stream URL, and Motion 4.3 no longer knows `rtsp_transport`, as the log showed. Please open a new issue with `motion.log` if that URL still fails with a current release.

closed 2026-11-11 by JamBalaya56562