motioneye-project/motioneye #2545: USB camera and the use of v4l2-ctl --set-ctrl horizontal_flip=1

evandene, 2022-06-28:
Hardware and OS: Raspberry pi 2W Zero, latest Raspberry pi OS.

I'm in the process designing a inspection USB camera. When the camera rotates 180 degree the image should flip. In my setup I'm trying a terminal command "v4l2-ctl --set-ctrl horizontal_flip=1" but that doesn't work. The system comes with "unknown command" 'horizontal_flip'
Can somebody help this oldy solving the problem?
If I can get this to work, I would like to get this command under a tactile button.
[image]

assetkad, 2024-02-26:
did you solve it?

evandene, 2024-02-26:
Yes we did solve the problem by upgrading the firmware of the camera