motioneye-project/motioneye #2551: Spinup hdd and s.m.a.r.t. read when I open motioneye

Nuke142, 2022-07-01:
I use _https://hub.docker.com/r/djaydev/motioneye/_ docker on UNRAID OS. All paths mounted to NVMe disk.
When I open motioneye website to watch my camera - both my HDD start spining, in syslog i see that motioneye read SMART.
This is bad. 99.9% of the time my hdd's should sleep.
That happens not each time. But each ~12 hours i guess.

MichaIng, 2022-07-03:
Could you verify whether it's the same with the new `dev` branch base Docker container: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

Nuke142, 2022-07-04:
Yes, same problem. I tested few hours after install `dev` branch - no spin-ups.
After 10+ hours i access my camera again and it takes about 10 seconds because of disk spin-ups.
[image]

Nuke142, 2022-07-18:
I think problem only with my case. Disabled all plugins and dockers and can't repeat that problem. No spinups anymore.
I add some info if i investigate it further.

closed 2022-07-18 by Nuke142