motioneye-project/motioneye #2556: Error : Format not supported or inaccessible/unsuitable for playback

zaphod24, 2022-07-10:
I have been using wansview Q5 cameras for a few years now with MotionEye. I recently added a few more cameras and it seems that the newer Q5's from Amazon now have H265 as the rtsp codec instead of H264. When trying to view these with the embedded video player in MotionEye, it gives this error:

Error : Format not supported or inaccessible/unsuitable for playback.

If I download the file, it plays fine with VLC. Is there any way to add H265 support to the embedded video player so I don't have to download the videos?

zagrim, 2022-07-11:
AFAIK it is about the browser having suitable video decoder available. Please try another browser (desktop if you've only tried mobile so far), and to decrease the Movie Quality setting in ME (say, 80% instead of 100%, or something like that), it might help, too. Or if you are using Movie Passthrough then you'd need to  tune the video setting in the camera UI instead.

zaphod24, 2022-07-19:
Cutting the video quality down to 80% or less and forcing the new cameras with HEVC not to use pass through did the trick

JamBalaya56562, 2026-11-07:
Closing as solved, as you confirmed: lowering the movie quality to 80% or less and not using passthrough for the HEVC cameras made the movies play.

closed 2026-11-07 by JamBalaya56562