motioneye-project/motioneye #2559: Very high load, low FPS, etc

blacc01, 2022-07-15:
Hi,

I successfully could install MonitonEye software on a Bullseye Lite (no desktop) using Raspberry Pi4 B 4GB version.

I attached 6 cam's as below:
[image]

And here the top output. You can see i have a load of about 10-11 which is too much:
[image]

What settings should i change to lower the system load, and get higher live fps (a separate monitor is installed to see all the cams always).

Here the config of cam1:
[image]
[image]
[image]
[image]
[image]

Thanks a lot!

Pete123123, 2022-07-15:
Maybe try lowering Video resolution and Movie and Picture quality and see what is acceptable. I would also change movie format to mpeg (1st selection).  I tried adjusting these settings for my RPI Zero 2W because the video was lagging and it did speed things up eventually.

Can I ask why you have a schedule from 0:00 - 23:59?

blacc01, 2022-07-15:
@Pete123123 thanks, lowering resolution and quality had a little helped, but not much. System load is now about 6-7, which is high. Another problem is that the fps goes always after some minutes to 0.0/0.0 when the focus is not on that browser tab. So its impossible to watch on another monitor the cam's. Maybe 6 cam's too much for the RPi, i don't know. I'm using a linux server (i5-2500) with Bluecherry for the 6 cam's without any problem. I wanted to swap to RPi, as the RPi power need is about 4,5W, and the i5-2500 45 W.
Schedule from 00:00 to 23:59 is because i need to record events all the day.

Pete123123, 2022-07-15:
Other ideas: 
1) Maybe disable working schedule. Mine is off and I have it recording 24/7. It seems like extra conditions for OS to check.
2) I put a cron script to restart RPI every 24 hours. Only takes about a minute.
3) How long is your usb cord to RPI? I first made a 20 foot long usb cord with 22 gauge wire and had weird performance issues. I guess voltage drop was too much for CPU to handle. When I shortened it to about 3 feet, things got a lot better

blacc01, 2022-07-15:
> Other ideas:
> 
> 1. Maybe disable working schedule. Mine is off and I have it recording 24/7. It seems like extra conditions for OS to check.

So, that setting is nothing todo with the recording schedule? What is that for?

> 2. I put a cron script to restart RPI every 24 hours. Only takes about a minute.

Thats a good idea, thanks.

> 3. How long is your usb cord to RPI? I first made a 20 foot long usb cord with 22 gauge wire and had weird performance issues. I guess voltage drop was too much for CPU to handle. When I shortened it to about 3 feet, things got a lot better

I have only one USB cable, which is for the external HDD to store the events. ALl my cameras are IP cameras.

Pete123123, 2022-07-15:
1) I think it's for stopping camera at certain times of the day. I don't use it so I just disabled the whole schedule.

Nuke142, 2022-07-18:
Change movie quality to 25%. My camera can do 8mbps, i thought that 100% in motioneye = 8 mbps but it is not. 100% is xxx mbps and VERY big file size. So i stoped at 25% = ~2,5mbps. Small sizes, low cpu usage, stable 25fps.

Nuke142, 2022-07-18:
@blacc01 Change movie quality to 25%. My camera can do 8mbps, i thought that 100% in motioneye = 8 mbps but it is not. Motioneye eye does not know much bitrate and quality your camera can handle. So 100% in Movie Quality is xxx mbps and VERY big file size (5sec = >500MB). So i stoped at 25% = ~2,5mbps. Small sizes, low cpu usage, stable 25fps.

JamBalaya56562, 2026-11-12:
Thanks to Pete123123 and wpresident for the answers. Six IP cameras on a Pi 4 means Motion decodes six streams in software, so the load mostly comes from the camera resolution and frame rate. Lowering the resolution and Movie Quality, as suggested here, helps. If your cameras have a substream, use it in motionEye: that lowers the load further. Closing as answered; please open a new issue with your versions if it is still a problem on a current release.

closed 2026-11-12 by JamBalaya56562