motioneye-project/motioneye #2560: No auto exposure with OV5647 at high resolutions

jrhansen, 2022-07-16:
Hi

MotionEye and Pi camera newbie here :)

I bought a bunch of OV5647 cameras and Raspberry Pi Zero 2 W's, and it seems that the auto exposure function seems to stop working (the exposure doesn't change according to overall light level) when I set the resolution to anything over 1280x720. Is this a known "feature" with the particular camera or MotionEye?

In general, I don't understand the camera settings that is shown in MotionEye (Auto Exposure, Auto Exposure Bias, etc.), because they don't seem to do anything. Also, when using the v4l2-clt command the settings are not synchronized to the ones in MotionEye. Can anyone explain the relationsship between the camera settings in MotionEye and the ones listed by the v4l2-clt command?

Finally, the camera is connected to the CSI connector of the Raspberry Pi, but I can only add it to MotionEye as a "Local V4L2 Camera". Is that ok, or is this related to some of my problems?

Thanks in advance
Hansen :)

harakka, 2022-08-06:
I suspect this is related to the new V4L2 based driver and software stack in Bullseye. I've seen it talked about elsewhere but haven't come across any real solutions. The same module works fine with MotioneyeOS that still uses the old driver stack.

harakka, 2022-08-06:
I suspect this is related to the new V4L2 based driver and software stack in Bullseye. I've seen the problem discussed elsewhere but haven't come across any real solutions. The same camera module works fine with MotioneyeOS that still uses the old driver stack.

MichaIng, 2022-08-06:
> AFAIK the MMAL driver isn't available any more in newer distros like Bullseye

It is, but legacy display drivers need to be enabled, e.g. via `raspi-config` or by removing the `vc4-(f)kms-v3d` overlay from `/boot/config.txt`.

Basically it is all discussed here: #2425

mattlazarowitz, 2024-08-25:
I'm having the same issue and I've been trying to find a solution for weeks.
I've tried both the local V4L2 camera->mmal service 16.1 and Local MMAL Camera->VideoCore Camera
In the check I just ran, it's underexposed when set up as a Local MMAL camera at 1080p.  At 720p the exposure might be a little high, but the image quality is still good enough to use. 
When running as a V4L2 camera, the image at 1080p is completely overexposed. At 720p things work and I think the image quality is a little better than wuen set up as a local MMAL camera. The additional video controls I get with the V4L2 system also make this system preferable.

I did try to follow the libcamera threads to try using that but it sounds like I need to compile motion from scratch with a specific patch and possibly MotionEye as well. I'd be willing to give that a try with clear instructions and some confidence that the patches aren't totally stale.

mrneutron42, 2024-09-24:
I am also seeing this overly bright exposure when using autoexposure with a Raspberry Pi HQ camera at 1280x1024, using V4L2 camera setting, Raspberry Pi OS Bullseye July 2024, motionEye Version 0.43.1b2, Motion Version 4.7.0.
This behavior was not present when using the MMAL Camera setting in the 2020 version of MotioneyeOS.
It appears that the autoexposure is in dormant mode?  The exposure appears stay at whatever exposure that is arrived at when the Motioneye software starts.  If you manually change any exposure setting and click Apply, the autoexposure becomes active for a few seconds and resets the exposure to a reasonable level, and then goes dormant again.

JamBalaya56562, 2026-11-12:
This is the legacy camera driver: `bcm2835-v4l2` uses its still capture port above `max_video_width` x `max_video_height`, 1280x720 by default, and the exposure does not keep adjusting there. Raising the limits with `options bcm2835-v4l2 max_video_width=2592 max_video_height=1944` in `/etc/modprobe.d/` and a reboot keeps it on the video port. With the libcamera stack, the camera is used through `libcamerify` instead (#2425). Closing, as the exposure is the driver's.

closed 2026-11-12 by JamBalaya56562