motioneye-project/motioneye #2561: Motioneye - Cant get the streaming URL

AnBad2, 2022-07-17:
Hi,

I have installed MotionEye to my Raspberry 4. It is running DietPi. I am using a  USB-Cam from Logitech.

My Problem:

When I click at the following links within the configuration of motioneye/camera:
[Snapshot URL](javascript:showSnapshotUrl())
[Stream-URL](javascript:showMjpgUrl())
[Eingebettete URL](javascript:showEmbedUrl())
no Tab or Windows opens, no link is shown.
What is wrong? How to I get the streaming URL?
Kind regards.

caldir65, 2023-09-19:
Hi,

same here in Motioneye ported to phyton 3.x

motionEye Version | 0.43.0
Motion Version | 4.5.1
Betriebssystemversion | Raspbian 11

Greetz

RobertKyuH, 2024-12-01:
The same issue, how to fix it?
Thx

closed 2026-02-21 by Marijn0