motioneye-project/motioneye #2562: Cameras randomly stop detecting motion

zaphod24, 2022-07-19:
I have been using Wansview Q5 cameras in addition to RPi0W (running motioneyeOS). As electrical storms have slowly taken out my RPi0W cameras (and because supply chain issues make them VERY expensive to replace), I have replaced everything with Q5 cameras. After changing over the last camera (I run 7 total atm, only 3 are setup for motion and recording), I am noticing that one or more cameras now randomly stops detecting motion. I can still view the camera feed on the motioneye web page and the Q5 has embedded time which is current, but no motion events and recording is triggered.

I am running dev motioneye on an Odroid N2 with Ubtunu 20.04.4. CPU usage is usually around 400% and there is around half of the total memory free or available for buffers.

Any idea what is causing motion to stop detecting? If I restart the motioneye service, everything works again for a while.

cryptage21, 2022-07-20:
I think I have the same problem.
Videos are missing randomly for my 5 cameras.

In log files I only have : 
```Jul 20 23:28:17 motioneye motion: [8:nc8:Camera 4] [ERR] [NET] netcam_rtsp_handler_reconnect: norm: Reconnecting with camera....
Jul 20 23:34:17 motioneye motion: [9:nc9:Camera 3] [ERR] [NET] netcam_rtsp_handler_reconnect: norm: Reconnecting with camera....
``` 
But I don't think that's the root cause because it concerns only 2 cameras while my 5 cameras have the problem. 

I don't know for how long I have this but it's clearly related to Python3 release and that's really annoying.

zaphod24, 2022-07-20:
I was seeing 400+ percent CPU utilization, so I've decided to cut it down to the 3 outside cameras I actually want alerts for, for now, and see if that eliminates the issue...

cryptage21, 2022-07-22:
That's not my case, no important CPU usage.

zaphod24, 2022-07-25:
Looks like mine may have been the result of the newer Q5 cameras changing from H264 to H265/HEVC. I wasn't detecting motion, just using the overview to monitor some rooms with the newer Q5s. I haven't had random motion detection stops for several days since removing the newer Q5 cameras from motioneye.

cryptage21, 2022-07-25:
Which motion version do you have ?
 I reverted back to Motion 4.3 from 4.4 (I didn't see before there were problems with this version).

It looks better... Will check in the next days.

zaphod24, 2022-07-27:
4.3

MichaIng, 2022-10-09:
I'm about to add motion 4.4 support: #2462

MichaIng, 2026-02-12:
I'm closing this issue since motion 4.4 has been added a while ago. But re-reading the issue, was motion 4.4 actually the solution or not the issue? Is it still an issue with latest motion? Feel free to reopen, if this is still the case.