motioneye-project/motioneye #2563: How to install MotionEye correctly?

jrhansen, 2022-07-21:
Hi

I am getting increasingly frustrated with/confused about how to get a CSI camera (OV5647) on a Raspberry Pi Zero 2 W to work properly. I hope someone is able to answer some very basic questions 😃 

I have two RPi's running with cameras attached - let's call them A and B. I tried numerous ways of installing MotionEye on both of them to get the camera to work in MotionEye, and to be honest I lost track of what I did. The cameras work on both systems now, but:

System A: Auto exposure does not work with resolutions above 1280x720. Above that, the camera does not correct for under/over exposure at all. I want 1920x1080.
System B: The FOV is much smaller than on system A even though both systems have the same camera with wide-angle lenses attached to them.

I want a wide FOV AND auto exposure to work. I swapped cameras as well as SD cards, and the "symptoms" follow the SD cards, so they are software issues. So:

1. What is the correct installation method for MotionEye when using Bullseye Lite and a camera attached to the CSI port?
2. Should the legacy camera support be enabled or disabled in raspi-config?
3. Should `camera_auto_detect=1` go in /boot/config.txt or not?
4. Should the cameras be added as V4L2 or MMA cameras?
5. Is there something wrong with Bullseye in general with regards to CSI cameras?

I have now started over with a new installation of Bullseye Lite and then following this procedure:

https://github.com/motioneye-project/motioneye/tree/dev#installation

... but I can't get MotionEye to connect properly to the camera regardless of the combination of the parameters (legacy camera enabled/disabled etc.). The picture in MotionEye is either a camera with a cross over it, or a light gray picture with the text "Unable to open video device" (or something like that).

I tried following these different installation methods:

https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye
https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian
https://github.com/motioneye-project/motioneye/tree/dev#installation

Please, help 😢

Baenwort, 2022-11-30:
@jrhansen Did you solve this? I'm hitting the same issue with the dev branch, bullseye, and a CSI camera.

MichaIng, 2022-11-30:
Generally, with legacy camera stack, it should be MMAL camera, with modern camera stack, it should be V4L2 camera. But the latter doesn't work good either. Correct modern RPi DSI camera support (`libcamera`) has to be added: #2425

However, enabling legacy camera stack and adding it as MMAL camera should work. If not, please check:
```sh
vcgencmd get_camera
```

closed 2026-11-11 by JamBalaya56562