motioneye-project/motioneye #2566: [Feature Request] Sound in the webinterface

Xitee1, 2022-07-23:
I know there are already a few issues open related to sound, but none of these do match my expection.

I want to be able to click on a camera (or maybe a sound-icon in a corner of the camera image) which will then also output my camera's sound (Reolink Network Camera connected over RTSP). So basically like the Reolink PC Client, but in the MotionEye webinterface.

coder0107git, 2022-07-24:
Unfortunately with the way motioneye works right now it would be very complicated to do. Currently motioneye only sends images continuously to the web interface to give the illusion of video. To add audio motioneye would need a major rewrite.

Xitee1, 2022-07-24:
Okay that is pretty sad. Do you by any chance know a software that can provide video and audio from a RTSP Stream to a webinterface?
It doesn't need motion detection or anything like that because my Reolink cameras have that built in. I just want a way to reliably display my cameras on a webinterface. I don't want to use the Reolink Cloud Service and Home Assistant sadly isn't so reliable with the Cameras and often hangs and has sometimes long loadi g times even when I'm in the same Network.

developerfromjokela, 2022-09-26:
@coder0107git current video stream in the UI is very inefficient. Proper MJPEG or RTSP stream would be a better option.

closed 2023-05-02 by Xitee1