motioneye-project/motioneye #2567: Save image on motion detetection

meoluc, 2022-07-23:
Sometimes not save the image saved on motion detection. I save images on internal SSD. 
I send message in telegram group, the message arrive when motion detetection Is activated, but don't send the ultimate image....

zagrim, 2022-07-24:
So you get the notification with no image attached? At least with email notifications that is a known issue when some conditions are met:
1) If there are a lot of media files, the scan of the files could take too long. In this case there should be a clear indication in the log (at least for email: "ERROR: timeout waiting for the media listing process to finish"), but if you are using SSD for storage that might not be the case (unless there's a ton of media files).

2) I'm not 100% sure of this, but based on my own experience I suspect that at least if the Maximum Movie Length if long enough, the still image might not yet be saved when the notification is being sent.

Please see here for more detailed discussion (especially on 2): https://github.com/motioneye-project/motioneye/issues/1956#issuecomment-734061791

If you are not able to sort it out, please do attach log file on a timespan when motion is being detected and notification being sent.

DeadlySin2, 2023-03-07:
The problem is that i get no notifications, and on the link attached people receive notifications without images/

free-cup-holder-exe, 2026-11-08:
The listing timeout zagrim describes is mostly gone in 0.45.0b1: #3425 (Telegram) and #3414 (email) make the senders list only the pictures within the attached-images time instead of the whole day folder, so a big folder no longer costs the picture. If a notification still has none, no picture was saved inside that time span: still images need a capture mode other than Manual, and with interval snapshots the span has to cover at least one interval.

free-cup-holder-exe, 2026-11-08:
Closing as completed; please reopen if a notification still arrives without its picture on 0.45.0b1 with still images enabled.

closed 2026-11-08 by free-cup-holder-exe