motioneye-project/motioneye #2569: Webinterface

Max6025, 2022-07-26:
I would like to make some changes to the web design and I would like to know where the web server data is stored.

Eldon-498, 2023-09-21:
HI, did you figure this out? Am also trying to make changes to the web design.

JamBalaya56562, 2026-11-12:
The web interface is part of the Python package: the HTML is in `motioneye/templates/` and the scripts, styles and images in `motioneye/static/`, in the package directory of your install (e.g. `/opt/motioneye/lib/python3.*/site-packages/motioneye/` with the current installer). Changes there are overwritten on every update, so keeping them in a fork of this repository and installing from it is easier. Closing as answered.

closed 2026-11-12 by JamBalaya56562