motioneye-project/motioneye #2574: Dumb user issue: how to make a pi3b+ run on ethernet

Jens5395, 2022-08-02:
So I have a pi3B+ that has been running motioneye for more than a year via wifi. I wish to transition it to running on ethernet. Silly me thought that I could just switch wifi off in the configuration screen, hook up an ethernet cable and be off to the races .....WRONG
What I found on the web indicates that the pi needs to have the ethernet cable plugged in on the first boot but surely therew must be a way of getting ethernet to work without reformatting the sd card and starting from scratch.

Could I get a hint on what the proper procedure is for this switch?

Jens5395, 2022-08-02:
A related question: I have switched wifi off in the configuration but wish to revert, at least for the time being, back to wifi. Of course the pi doesn't come onto the network properly and it is headless. Which configuration file contains the switch for wifi so I can change it directly on the SD card? I can't seem to find it.

Jens5395, 2022-08-02:
Got it solved ... I assumed that since I had assigned a static IP fir the pi, that the wired setup would use the same IP. Turns out it doesn't. Once I went to the right IP everything was running fine.

zagrim, 2022-08-03:
Good for you, then :+1: I guess this can then be closed.

closed 2022-08-03 by zagrim