motioneye-project/motioneye #2579: Motion detected push notification

Saku241, 2022-08-17:
Hello !
I would like to know from scratch how can I make push notification on my phone with the picture taken from motion 

Is it possible ? I would really like to know how
Thx !

MichaIng, 2022-08-17:
Many thanks for your request.

For a push notification, if you don't mean one send by browser while the motionEye web UI is opened, one would need a phone app and use a 3rd party service, e.g. [Firebase Cloud Messaging](https://firebase.google.com/docs/cloud-messaging/) on Android. So I think this is out of our scope. But sending a push notification through the browser should work, while I'm not sure whether this is sufficient benefit.

Best replacement currently is the sendmail feature.

lowlyocean, 2022-08-17:
@Saku241 I wonder if you can use webhook feature in motioneye to achieve "push notification" on your phone. I'm trying to do the same thing with a combination of [endpoint in Node-RED](https://cookbook.nodered.org/http/create-an-http-endpoint) and [these Alexa nodes](https://flows.nodered.org/node/node-red-contrib-alexa-remote2-applestrudel) . Unfortunately the webhook feature seems to be broken as I reported in issue https://github.com/motioneye-project/motioneye/issues/2573

MichaIng, 2022-08-17:
Webhook should work with development version: https://github.com/motioneye-project/motioneye/tree/dev#installation

Saku241, 2022-08-17:
I already achieve to get push notification with webhook and ifttt app but I was wondering to have a picture attached to the notification with the motion just like mail notification

MichaIng, 2022-08-17:
Right, with the webhook feature no image can be transferred, unless you can do some custom scripting at the webhook server and images are uploaded to a server where the webhook has access to.

JamBalaya56562, 2026-11-13:
motionEye 0.43.1 added *Send Telegram Notification* under *Motion Notifications*: with a Telegram bot set up, it sends the motion pictures along with the message, and the Telegram app shows it as a push notification on the phone. Pictures need *Still Images* on.

closed 2026-11-13 by JamBalaya56562