motioneye-project/motioneye #2580: Why install instructions do not mention pip2?

Basilisvirus, 2022-08-18:
I see that alot of installation problems are solved by using pip2 to install motioneye.
Instead, the installation instructions: https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian only mention "pip". 

Why the installation instructions show pip instead of pip2?

MichaIng, 2022-08-18:
Those instructions are basically outdated, I guess from a time where `pip` was still `pip2` while nowadays Python 2 is rarely even natively available on distros.

The new development version is Python 3 based and has consistent install instructions across all distros (aside of probably Arch which has own AUR packages): https://github.com/motioneye-project/motioneye/tree/dev#installation

Basilisvirus, 2022-08-19:
Oh I see. 
I am trying to install motioneye on raspbian bullseye, and now I see the correct installation notes: https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye Thanks!!

closed 2022-08-19 by Basilisvirus