motioneye-project/motioneye #2582: Third version of wifi chip Rpi0 2 W using motioneye Jaspers image

fils02, 2022-08-29:
Hi @jawsper, hi @starbasessd I found your build for motioneyeos rpi0 2 w, and I tried to run but it seems not connecting to wifi even if I tried a lot of times with different wpa_supplicant files. Then I found this topic about the two version of chip and realized that mine is a third one version!
I was wondering if you could add a new version for this, it'll be very helpful for me!

chip serial RP3A0-AU
2104
201221A

Thank you very much in advance 

#2265

JPS239, 2023-08-03:
Same problem here, tried all available images, but without any luck, not connecting to wifi :(

chip serial RP3A0-AU
2318
230413A

Maybe it's only a small fix (for someone with the necessary knowledge...) Thank you so much in advance!

Slipery, 2023-10-10:
Same problem here, tried all available images, but without any luck, not connecting to wifi
chip serial RP3A0-AU
2320
230413B
[image]

JamBalaya56562, 2026-11-12:
This is about the Wi-Fi driver in the motionEyeOS image, which is not part of motionEye and cannot be fixed here. The same problem is tracked in motioneye-project/motioneyeos#3065, so please follow it there. Without motionEyeOS, Raspberry Pi OS with motionEye installed as in the README uses the stock Wi-Fi support for all chip revisions. Closing in favour of that issue.

closed 2026-11-12 by JamBalaya56562