motioneye-project/motioneye #2583: Cameras previews don't fit to full browser window

lowlyocean, 2022-08-29:
With 4 cameras, I expected Layout Columns = 2, Layout Rows = 2 and Fit Frames Vertically enabled would make all previews scale to fit into equally sized quadrants as I resize the window. Is there anyway to achieve that behavior? I find myself having to scroll even when F11 my browser into full screen mode, because the bottom of the 2nd row doesn't fit the window.

zagrim, 2022-08-30:
Only Layout Columns does actually squeeze the frames into the available space, Layout Rows doesn't do that and that can be seen as a bug. What Fit Frames Vertically switch does is to enable Layout Rows and basically the vertical size control, so in principle you are correct with your assumption that it should work.
What is your screen resolution and the camera frame sizes?

lowlyocean, 2022-08-30:
My screen resolution is 1920x1080. The first 3 cameras are 352x240, and the 4th camera is 320x240

lowlyocean, 2022-09-07:
Mind reviewing MR? Tested with 4 cameras, various combinations of rows/columns renders as expected (with and without settings panel active, both >1200 and <1200 window width). There's no inherent reason for it to not work with different # of cameras.

closed 2026-10-29 by MichaIng