motioneye-project/motioneye #2586: Bullseye, motionEye Version 0.43.0, Motion Version 4.3.2, working schedule not working

bulgin, 2022-08-31:
Hello.  Could use some advice.  I've set a working schedule but motioneye zips right past it and continues to operate during off schedule hours.  I've checked the clocks on the raspberry pi and the time presented in the screens of motioneye and they are all in sync and accurate for this time zone.  Can't figure out whats's wrong.

Any help much apprecicated.
[image]

zaphod24, 2022-09-02:
I've noticed the same with my setup ever since I updated to the python3 version. It just records all the time, not during the scheduled times.

zaphod24, 2022-09-28:
No ideas?

MichaIng, 2022-09-28:
Can you check whether `journalctl -u motioneye` contains any errors?

MichaIng, 2023-04-15:
Let's keep this issue tracked here: #2663
I assigned a milestone as well.

closed 2023-04-15 by MichaIng