motioneye-project/motioneye #2597: Performance bad since some time

speeskeek, 2022-09-20:
Hi all. Been using motioneye for >5 y now i guess. Very happy with it. 
since a week (?) I have very bad performance of the streams. Actually - it are still images instead :)

when I disable/delete all the cams but one it works again. 

Before it worked perfectly. 
cpu usage is same as bedore. Where can I look? 
thanks. 
using motioneye 0.42.1 on a odroid hc.

MichaIng, 2022-09-20:
If/as there was no update of motionEye in the meantime, it is out of our control but a network or server issue/change.

However, there is actually a new Python 3 based development version which we want to pass as pre-release soon. Here is how you can install it: https://github.com/motioneye-project/motioneye/tree/dev#installation

speeskeek, 2022-09-20:
I’ll try the python3 version. 

Inguess some dependency was updated?

MichaIng, 2022-09-20:
Babel und boto3 have been added as dependencies, since it is localised now and allows S3/AWS uploads. And of course the modules need to be installed for Python 3. But there are no specific version requirements for dependencies.

speeskeek, 2022-09-20:
I installed the python3 version. But it’s the same. 

Somewhere there is a problem … becouse with resolution dimmer set to 10% it works with 2 strams.

speeskeek, 2022-09-20:
“”I guess some dependency was updated?””

i meant: maybe a dependency (motion or librurbojpeg, …) has been updated recently and is the origin of my prroblem

speeskeek, 2022-09-20:
But when running the python3 dev version I still see in htop: python2 /usr/local/bin/meyectl startserver

coder0107git, 2022-09-20:
The python2 motioneye startup service probably is still enabled. I think it can be disabled by running `sudo systemctl disable motioneye`.

MichaIng, 2022-09-20:
The Python 3 service has this name. Use `systemctl` to check all active services to find the one responsible to start the Python 2 version of motionEye.

To verify that `motioneye.service` is the Python 3 one:
```sh
systemctl cat motioneye
```

speeskeek, 2022-09-21:
`root@HomeCloud:~# systemctl cat motioneye
# /etc/systemd/system/motioneye.service
[Unit]
Description=motionEye Server

[Service]
ExecStart=python2 /usr/local/bin/meyectl startserver >Restart=on-abort

[Install]
WantedBy=multi-user.target
`

the systemd file still sais python2. If I change the 2 to 3 (so I get python3) it doesn work :(

MichaIng, 2022-09-21:
Did you run `sudo motioneye_init`? It overwrites this service.

speeskeek, 2022-09-21:
(thanks for the help already!)

yes I did the init script. finished with no errors

MichaIng, 2022-09-21:
Ah sorry, we made it conservative and do not overwrite the file if it exists already. Please remove it first:
```sh
rm /etc/systemd/system/motioneye.service
sudo motioneye_init
```
Probably we should install the new one but create a backup of the old one.

speeskeek, 2022-09-21:
okay - getting there I guess

> systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-09-21 08:11:16 CEST; 29s ago
    Process: 28690 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=255/EXCEPTION)
   Main PID: 28690 (code=exited, status=255/EXCEPTION)

Sep 21 08:11:14 HomeCloud systemd[1]: Started motionEye Server.
Sep 21 08:11:15 HomeCloud meyectl[28690]: configure_logging cmd motioneye: False
Sep 21 08:11:15 HomeCloud meyectl[28690]: configure logging to file: None
Sep 21 08:11:15 HomeCloud meyectl[28690]: CRITICAL: pid directory "/var/run" does not exist or is not writable
Sep 21 08:11:16 HomeCloud systemd[1]: motioneye.service: Main process exited, code=exited, status=255/EXCEPTION
Sep 21 08:11:16 HomeCloud systemd[1]: motioneye.service: Failed with result 'exit-code'.

MichaIng, 2022-09-21:
Please set `run_path /run/motioneye` in `/etc/motioneye/motioneye.conf`. It runs as unprivileged user, hence has no privileges to write to `/run` directly. The sub directory is created by the systemd unit.

speeskeek, 2022-09-21:
okay .. I had a few more write-permission troubles but tthey are resolved. first impressions are, the choking is gone. running now smooth with 3 cams on. I'll add the others one after one.....

closed 2022-09-21 by speeskeek

MichaIng, 2022-09-23:
Great to hear 👍.