motioneye-project/motioneye #2599: RPi Zero W + Docker - pillow build error

RamonAbudAlcala, 2022-09-22:
I am having this #2495 same output (of pip failing to build pillow) while building the most recent Dockerfile in the `dev` branch. I'm running this on a fresh install of raspberrypiOS bullseye 32bits on a raspberry pi zero w. 

I ran the following command from the root folder (motioneye) of a clone of this repo.
```
docker build \
--build-arg VCS_REF=$(git rev-parse HEAD) \
--build-arg BUILD_DATE=$(date +"%Y-%m-%dT%H:%M:%SZ") \
-t motioneye-project/motioneye:dev \
-f docker/Dockerfile .
```

Any help is appreciated,

MichaIng, 2022-09-22:
Most likely some library is missing. Can you show the exact error message?`

Our last build from two days ago went well: https://github.com/motioneye-project/motioneye/actions/runs/3091683618/jobs/5002058461

You are aware of our Docker images, right? https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

JamBalaya56562, 2026-11-10:
Closing, as this got no answer with the exact error. Building pillow on a Pi Zero W needs its build libraries, while the prebuilt images at `motioneyeproject/motioneye` (and ghcr.io) avoid the build. Please open a new issue with the full log if building still fails.

closed 2026-11-10 by JamBalaya56562