motioneye-project/motioneye #2601: next release

bigmac5753, 2022-09-24:
is there an ETA for the next release?
Is there a beta version available for docker?

MichaIng, 2022-09-24:
Docker dev images are available here: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

I want to do a beta to PyPI as well. @jmichault you have best knowledge about how to do that: A pre-release to PyPI is what I suggest.

MichaIng, 2022-12-23:
PR up for PyPI upload workflow: #2675

JamBalaya56562, 2026-11-11:
Closing, as releases are out since then: the PyPI workflow from #2675 publishes them, 0.44.0 is the current release and 0.45.0b1 a pre-release (`pip install --pre motioneye`), and the Docker image is on Docker Hub as `motioneyeproject/motioneye`.

closed 2026-11-11 by JamBalaya56562