motioneye-project/motioneye #2603: WARNING: mjpg client connection for camera 3 on port 8083 is closed

pickonedev, 2022-09-25:
Hi!

When I add the 3rd camera (does not care what kind of camera), I get this error "WARNING: mjpg client connection for camera 3 on port 8083 is closed". Any idea what can I do to solve this?
The camera feed it is not shown.

Thank you!

MichaIng, 2022-09-25:
The `lsb_release` error is not related and can be ignored, but also solved by installing the `lsb-release` package, e.g. on Debian/Ubuntu:
```sh
apt install lsb-release
```

Which OS do you use, and which motionEye version? If 0.42.1, please try to use the new Python 3 based dev version, which has quite a bunch of bugs solved: https://github.com/motioneye-project/motioneye/tree/dev#installation

pickonedev, 2022-09-25:
I am using:

 Home Assistant 2022.9.6
Operating System 9.0

With MotionEye Add-On 0.18.0

MichaIng, 2022-09-25:
Ah okay, I have no experience or insights into the Home Assistant OS, but it uses the older motionEye for sure. We really need to get v0.43.0 released, so distros and other appliance developers can update theirs 🤔.

JamBalaya56562, 2026-11-10:
Closing, as this was with the Home Assistant add-on 0.18.0, which shipped an older motionEye, and got no further reports. Please open a new issue with `motion.log` if a third camera still loses its stream with a current release.

closed 2026-11-10 by JamBalaya56562