motioneye-project/motioneye #2607: Droped frames

zajaaczki, 2022-09-30:
I have 2 instalations at home HP T620 5 cameras and office old server intel xeon processor with 2 cameras.
In both locations recordings I see droped frames, taht means someone walking and disapeard.
Different settings and different cameras, wifi and lan connections TCP not UDP.
Some bugs
motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Linux 5.10.0-17-amd64

MichaIng, 2022-09-30:
Can you test whether it works better with the new Python 3 based motionEye version? https://github.com/motioneye-project/motioneye/tree/dev#installation

zajaaczki, 2022-10-03:
I will try.
Can I use upgrade option or full reisntall is required?

JamBalaya56562, 2026-11-10:
Closing, as this got no follow-up after the suggestion to try the Python 3 release, and was with motionEye 0.42.1. Please open a new issue with `motion.log` if movies still drop frames with a current release.

closed 2026-11-10 by JamBalaya56562