motioneye-project/motioneye #2609: What is the max password length allowed?

matttb1984, 2022-10-01:
I have run into this a couple times now with both MotionEyeOS (20200606) and now with MotionEye (0.42.1) running on Bullseye, both on RP4s.

It seems if you enter a password that is too long (not sure what the threshold is) you then cannot ever login again.  Just tells you invalid credentials.  Have to reformat and install again fresh.

Would be great to know what the specs are for the password ie. length, special characters, ect.

I generally like to use long passwords for things randomly generated in the 20-30 character range, usually without any issue.

MichaIng, 2022-10-02:
Is it really the password length or could it be an issue with some special character? In the code I do not see a reason why it shouldn't work so far, it is stored as hash with fixed length anyway: https://github.com/motioneye-project/motioneye/blob/dev/motioneye/config.py#L739

But should be easy to test.

JamBalaya56562, 2026-11-10:
Closing, as since 0.44.0 (#3332) the passwords are stored as an argon2 hash, so their length and characters no longer matter: 30, 200 and special-character passwords all verify. Please open a new issue if a password still stops working after saving it with a current release.

closed 2026-11-10 by JamBalaya56562