motioneye-project/motioneye #2610: Add freebsd_init to extras please

paulprocacci, 2022-10-02:
Attached you'll find a freebsd_init.txt file.  (Should be renamed to freebsd_init but github wanted a file extension)

[freebsd_init.txt](https://github.com/motioneye-project/motioneye/files/9691862/freebsd_init.txt)

Please include this to your extras directory.

Appreciate it!

MichaIng, 2022-10-02:
I suggest to merge BSDish setup into the existing `linux_init` script (which can be renamed). Make the service a dedicated file and use the existing example config but adjust only the settings/lines which really must be different on BSD. That way we minimise doubled code.

paulprocacci, 2022-10-02:
A good suggestion.  That would mean linux_init would need to be rewritten to NOT depend on bash and instead lean on normal sh semantics.

MichaIng, 2022-10-03:
Ah, no bash on BSD? However, shouldn't be a big deal and has other benefits to be bourne shell compatible. I wanted to add some info output to the script anyway, probably `read -p` dialogs to give users more control about which steps are done. I can make bash => sh migration in the same turn.

paulprocacci, 2022-10-03:
The default and available shells on BSD are /bin/sh, /bin/csh and /bin/tcsh.
/bin/sh on BSD is an IEEE Std1003.1specification for the shell designed by POSIX many moons ago.
/bin/sh is guaranteed to be on all operating systems (even if linux, et al symlink them elsewhere).

bash can be installed, but then you're looking at additional and in my humble opinion uneeded requirements.  It installs under /usr/local/bin so in that case you'd actually have to ensure all scripts' shebang line read '#!/usr/bin/env bash' so as to PATH gets used.

To ensure it works everywhere, at least as many locations as humanly possible, a shebang line that reads '#!/bin/sh' and follows IEEE Std 1003.1 is a safe bet of working no matter where it runs.

MichaIng, 2022-10-03:
I'd actually use `#!/usr/bin/env sh` shebang nowadays, but both should work very reliably. However, I agree on all points. Also I want to invoke the script via `scripts` in setup.cfg instead of `console_scripts` as part of a Python function wrapper.

paulprocacci, 2022-10-03:
As for the other reference to #1616 I read through it and even though I understand it's trying to add support for FreeBSD I'm not actually effected by any of the problems it's trying to fix.

For example, sha1sum exists on FreeBSD and the command works as described without needing a patch.
command -v <item> <item> won't work on FreeBSD.  It allows only a single argument.  In this case, your suggestion would fail.

The whole disk section of detecting disks, I have no idea why motioneye is doing disk detection at all.  In the motioneye control panel, it's seeing the disk it's writing to and properly reporting on usage, but besides this I have no idea what its purpose is for.  I'm using zfs instead of ufs and perhaps that's the reason why it's working for me?  I'm not sure.

Baenwort, 2022-12-16:
Have these FreeBSD improvements made it into the dev branch?

closed 2026-11-01 by MichaIng