motioneye-project/motioneye #2611: Google Drive upload

LittleWenlock, 2022-10-05:
I have two Raspberyy Pi's running Motioneye with the same settings 
One is on Raspbian 10 (Buster), Motioneye 0.42.1 and everything is fine
Other is on Raspbian 11 (Bullseye), Motioneye 0.43.1 and everything is fine except nothing gets uploaded to Google Drive.

With "one", Test Service gives "Accessing the upload service succeeded!" and images upload when motion-triggered or manual.
With "other", Test Service gives ""Accessing the upload service succeeded! [object object]" and nothing uploads. Is this a permissions problem?

By the way, when you do "obtain key" with either version there's this warning from Google: **_"To help protect your account, Google will soon block apps that don’t comply with Google’s security policies. Make sure that you are using the latest version of this app. You can let the app developer ([ccrisan@gmail.com](mailto:ccrisan@gmail.com)) know that this app should stop using unsupported OAuth flows."_**

MichaIng, 2022-10-05:
Looks like this bug. I didn't find time to investigate yet: https://github.com/motioneye-project/motioneye/issues/2516

Tormund1985, 2022-10-19:
same problem here. Not uploading Google anymore.

ahlawahlaw, 2022-10-29:
Same here.  I can no longer obtain an Auth Key from Google.  Google is now blocking the app