motioneye-project/motioneye #2613: docker documentation query

SimonMcN, 2022-10-06:
Hi,
I'm following these instructions https://github.com/motioneye-project/motioneye/wiki/Install-In-Docker in order to try and manually build an up to date docker container (which I think is already being done now so I don't know if the documentation is out of date or not).

When I do a docker build, I am in the docker folder and the instruction copy . to /tmp.  But there is no motioneye source in . as I am in the docker folder.

Am I following the instructions incorrectly ?  I was not entirely clear on the extra docker file.

I appreciate you doing this project.  Looks nice :-)

MichaIng, 2022-10-06:
I suggest to use our new dev branch Docker image: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

For the old method, someone else would need to answer.

starbasessd, 2022-10-14:
I last modded the old instructions. 
What host OS, version of Docker (.io or-ce) and which set of intructions (Method 1, 2 or 3) did you use?
What version of container OS were you 'upgrading' to?
Were you still installing Python2.7 or were you trying to convert to 3.x? If converting, the new Devs will have to answer, and you want to use their image that Michalng posted...

JamBalaya56562, 2026-11-10:
Closing, as the build steps now live in `docker/README.md`: the build runs from the repository root with `-f docker/Dockerfile .`, so the source is in the build context, which is what was missing when building inside `docker/`. The prebuilt `motioneyeproject/motioneye` image is the easier way, as MichaIng suggested.

closed 2026-11-10 by JamBalaya56562