motioneye-project/motioneye #2614: Matter Support

leoheck, 2022-10-08:
Google is launching this Matter. I would like to know if the MotionEye project has any plan to support it.
We could integrate cameras with the Google Home app if support is provided.

Some relevant links:
https://www.theverge.com/2022/10/4/23386287/google-home-app-design-update-matter-features
https://developers.home.google.com/matter

zagrim, 2022-10-09:
What kind of support you are asking about? Being able to show Matter-supporting cameras in MotionEye (kind of like ONVIF support would work for ONVIF-supported devices), or using MotionEye as some sort of a bridge to connect non-Matter-supporting cameras to Matter ecosystem?

It took me quite a lot of clicks to end up to an article that says at least something concrete about how Matter is supposed to technically work (then again, it might have changed from that time already): https://www.theverge.com/22787729/matter-smart-home-standard-apple-amazon-google
But still, I'm not quite sure what it would even mean to "support" Matter. In my humble opinion, and as far as I understand it, that would anyway be quite far from the current reason for existence of ME. It might be a cool feature, though.

leoheck, 2022-10-09:
I am not sure too. But if motion/motion eye cameras could be integrated with the Google Home app it would be interesting.

MichaIng, 2022-10-09:
Wouldn't it be more reasonable to add the cameras/streams directly to Matter, rather than the altered stream piped through motion/motionEye? motionEye is an application to detect/record/notify about motion, not so much for providing a webcam stream for other applications. The latter would mean an unnecessary overhead of the processing done by ffmpeg, motion and in case motionEye.

In case, for providing camera streams as network cameras to remote applications, there are other applications with much less overhead, like [mjpg-streamer](https://github.com/jacksonliam/mjpg-streamer).

sonicpp, 2022-10-10:
> motionEye is an application to detect/record/notify about motion

To me, motionEye is more than that (but yes, it is where is the main focus). It is GUI for backing configuration, upgrading software...and I also for streaming camera. If I am not wrong, you can disable motion and use just camera streaming (but I agree, there would be still some overhead). Just like motionEye is using motion at the background, it is using some other apps at the background for streaming (theoretically it could use mjpg-streamer, if it does not).
I can of course manually install mjpg-streamer, configure it and use it from command line. But motionEye adds nice GUI for all the configuration around cameras.

leoheck, 2022-10-11:
The reasoning is that Google Home would provide a good interface to access cameras in the smartphone. It could be an extra feature if enabled since there are people that prefer to keep the installation offline. But since this conversation has started here I am thinking that it may not be an easy thing to add to this project here.

JamBalaya56562, 2026-11-15:
As discussed, motionEye is for detecting, recording and notifying about motion, not for passing camera streams on to other systems, and Matter would mean running a bridge for that. Adding the cameras to Matter or Google Home directly avoids the extra processing through motion and motionEye. Closing as not planned.

closed 2026-11-15 by JamBalaya56562