motioneye-project/motioneye #2616: Not clear how FPS mismatch b/w UI and (sub-)streams affect CPU usage

lowlyocean, 2022-10-10:
I am running on RPI so the CPU is constrained. Thus far, I've used following on all cams with success. Videos are recorded just fine, and the CPU usage is reasonable throughout.

```
Frame Rate in the UI set to 2
"Camera Device" is an RTSP substream that is low-res and 2 FPS 
"Extra options" includes netcam_highres pointing to RTSP main stream that is high-res and 30FPS
```

I just added a new cam ( different manufacturer, but configured same as above ) and it looks like it's not smoothly recording the main stream to disk after a motion event. But, if I change the Frame Rate in the UI to 30 FPS, the saved video is fine. The side-effect seems to be that CPU usage increases because there are 30 frames each second to perform motion detection on. Is that right?

Also, I noticed that mis-match of FPS in these 3 places changes how much video is recorded before/after the motion event. How is this expected to behave when the FPS isn't matched?

zagrim, 2022-10-12:
> I noticed that mis-match of FPS in these 3 places changes how much video is recorded before the motion. I'm seeing 25 or so seconds of video before the motion detection event even though I have "Captured Before" set to 1 frame

I have not myself done any serious testing of how Capture Before/After work with `netcam_highres`, but logically it should be so that the before/after settings refer to the frames used in motion detection. Myself I'm using `netcam_highres` but also `movie_passthrough` which probably might change things a bit. I seem to have some extra frames in the start and the end of captured videos, but I don't think it's really consistent and never as pronounced as in your example.

Anyway, UI "Frame Rate" corresponds to motion `framerate` which according to [docs](https://motion-project.github.io/motion_config.html#framerate) means: "Maximum number of frames to be captured from the camera per second. [...] Set this parameter to the maximum number of images per second that you want to store either as images or movies." But the docs are not really saying how that changes with `netcam_highres` (which in newer Motion has been renamed to [`netcam_high_url`](https://motion-project.github.io/motion_config.html#netcam_high_url)): "Motion detection on the normal resolution will trigger the saving of images from the high resolution stream."

So, there might be some issue in Motion on how it calculates how many frames of the highres video should be saved for each frame of the lowres video that is used for detection.

As for the other thing you mention I have no clue, but since AFAIK all the mentioned functionality relies solely on Motion, that could be a similar case. This sems like something that could be reported as Motion issue, but first you should verify that the camera config file content really is what one would expect based on ME UI, and preferably also do that for both 2 fps framerate and 30 fps framerate. There are instructions on how to run Motion with the config and how to report the issue here: https://github.com/motioneye-project/motioneye/wiki/Report-Motion-Issue

lowlyocean, 2022-10-12:
I noticed that I can also further complicate the configuration by specifying these in "Extra Options".
```
netcam_high_params framerate=30,capture_rate=30
netcam_params framerate=2,capture_rate=2
```
I checked the main stream for the existing cameras and this new camera in VLC - it looks like VLC's codec info doesn't report 30FPS from the new camera but it does for the old one. So maybe that's why I have to set 30 FPS manually in the UI for the new camera in order for it to record properly? If that's the case I wonder how VLC is able to render it without the FPS info - it must be something Motion ignores. So I was hoping the "extra options" above setting framerate & capture rate on the main/sub streams would fix things (it doesn't). 

Appreciate your suggestion - I do think motioneye is probably not contributing to the issue and may be better reported in Motion project.

JamBalaya56562, 2026-11-11:
Closing, as zagrim answered this: the Frame Rate in motionEye is Motion's `framerate` (`config.py:1091`), the number of frames Motion captures and checks for motion each second, so 30 there means 30 checks a second. How the frames captured before an event map to the `netcam_high_url` recording is Motion's behaviour, as you concluded too.

closed 2026-11-11 by JamBalaya56562