motioneye-project/motioneye #2618: Feature Request

starbasessd, 2022-10-16:
For any system there should be the equivalent, but for the Raspberry Pis I suggest the following:

Here is a quick script to log info:

>sudo nano /usr/bin/status.sh
copy and paste the following (minus the '- - - -')
- - - -
#!/bin/bash
temp=$(vcgencmd measure_temp)
volt=$(vcgencmd measure_volts)
throttled=$(vcgencmd get_throttled)
echo $(/bin/date) $volt $temp $throttled >> /var/log/info.log
- - - -
>[ctrl]x y [enter] to save
sudo chmod 755 /data/usr/status.sh
sudo crontab -e (select 1 for editor)

add the following:
>* * * * * /data/usr/status.sh
[ctrl]x y [enter] to save

It should now save the info once a minute to /var/log/info.log with a time/date stamp.

I've tested in RaspberryPiOS 32 & 64 bit.

MichaIng, 2022-10-16:
How is storing this info in some log file a feature relevant for motionEye? If at all, it would be an idea for motionEyeOS, but only if the info is at least also shown somewhere in the motionEye UI.

JamBalaya56562, 2026-11-15:
As @MichaIng said, this is only relevant for motionEye if it shows up in the UI, and that's already possible with a monitoring command. An executable `/etc/motioneye/monitor_1` (1 = camera id) like
```sh
#!/bin/sh
echo "$(vcgencmd measure_temp) $(vcgencmd measure_volts) $(vcgencmd get_throttled)"
echo 60 >&2
```
shows the values over the camera's picture and runs again every 60 seconds. Closing.

closed 2026-11-15 by JamBalaya56562