motioneye-project/motioneye #2620: No more Motioneye start possible after Bullseye update an re-install of Motioneye

Pedder007, 2022-10-21:
Hi all, I hope that someone can help.

I just updated my Raspi4B from Buster to Bullseye and afterwards re-installed Motoneye according to the manual, but I'm not able currently to get the server starting again.
While the installation there were no errors, but only some warnings about Python 2.7.

Now it looks like following:
````
pi@raspimotioneye:/ $ sudo systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-10-21 21:40:48 CEST; 14s ago
    Process: 12794 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 12794 (code=exited, status=1/FAILURE)
        CPU: 169ms

Okt 21 21:40:48 raspimotioneye systemd[1]: Started motionEye Server.
Okt 21 21:40:48 raspimotioneye meyectl[12794]: Traceback (most recent call last):
Okt 21 21:40:48 raspimotioneye meyectl[12794]:   File "/usr/local/bin/meyectl", line 5, in <module>
Okt 21 21:40:48 raspimotioneye meyectl[12794]:     from motioneye.meyectl import main
Okt 21 21:40:48 raspimotioneye meyectl[12794]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 28, in <module>
Okt 21 21:40:48 raspimotioneye meyectl[12794]:     import settings
Okt 21 21:40:48 raspimotioneye meyectl[12794]: ModuleNotFoundError: No module named 'settings'
Okt 21 21:40:48 raspimotioneye systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Okt 21 21:40:48 raspimotioneye systemd[1]: motioneye.service: Failed with result 'exit-code'.

````
and:
````
pi@raspimotioneye:/ $ apt-cache policy python-pip python3-pip
python-pip:
  Installiert:           (keine)
  Installationskandidat: (keine)
  Versionstabelle:
python3-pip:
  Installiert:           20.3.4-4+rpt1+deb11u1
  Installationskandidat: 20.3.4-4+rpt1+deb11u1
  Versionstabelle:
 *** 20.3.4-4+rpt1+deb11u1 500
        500 http://archive.raspberrypi.org/debian bullseye/main armhf Packages
        100 /var/lib/dpkg/status
     20.3.4-4+deb11u1 500
        500 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages
`````
and:
````
pi@raspimotioneye:/ $ python2 --version
Python 2.7.18
pi@raspimotioneye:/ $ python3 --version
Python 3.9.2
````

Pedder007, 2022-10-21:
Maybe the following as additional info:
`````
pi@raspimotioneye:/ $ sudo pip install motioneye
#DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: motioneye in /usr/local/lib/python2.7/dist-packages (0.42.1)
Requirement already satisfied: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye) (6.2.2)
Requirement already satisfied: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye) (2.11.3)
Requirement already satisfied: tornado<6,>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.1.1)
Requirement already satisfied: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye) (7.43.0.5)
Requirement already satisfied: MarkupSafe>=0.23 in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye) (1.1.1)
Requirement already satisfied: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.4.0.3)
Requirement already satisfied: futures in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.3.0)
Requirement already satisfied: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (0.5)
Requirement already satisfied: six in /usr/local/lib/python2.7/dist-packages (from singledispatch->tornado<6,>=3.1->motioneye) (1.16.0)
WARNING: You are using pip version 20.3.4; however, version 22.3 is available.
You should consider upgrading via the '/usr/bin/python2 -m pip install --upgrade pip' command.
`````
The outcome is the same, regardless I use 'pip' or 'pip2'.

coder0107git, 2022-10-21:
Try installing the new python3 version with the instructions [here](https://github.com/motioneye-project/motioneye/tree/dev#installation).

Pedder007, 2022-10-21:
Hi @coder0107git, thanks a lot for you quick hint! I directly tested it and it works 👍 
I only had to add 2 additional 'sudo's in section 2 ;-)

What I'm wondering about is, that I already did an installation in the same way 3-4 weeks ago on a test system and then it worked. However, Motioneye is now also live on my new Bullseye prod system, super 😄

MichaIng, 2023-03-30:
Seems to be solved, so I mark this as closed. Feel free to reopen if required.

closed 2023-03-30 by MichaIng