motioneye-project/motioneye #2621: port forwarded version seems to open motion

soapy75, 2022-10-22:
I am using raspberry pi 02 w. 

I have no problems with the local version with two cameras
[image]

but after portforwarding, the version shows a simplified version with seems to be motion, not motion eye.
[image]

I was presented with a log in pop up and managed to log (as admin)on the first time i opened it, but even after clearing cookies it no longer shows up. There is no admin or second cameras which seems to be a characteristic of the normal motion. I am still able to access the local version with no problems.

zagrim, 2022-10-23:
Which port did you forward? 8765 (or whatever you have when locally accessing ME in the first screenshot)?

soapy75, 2022-10-23:
Ohhhh the local port is 8765 but the forwarded is 8081, could that be the problem?

zagrim, 2022-10-25:
By default, port 8765 is the UI of MotionEye, 8081 is the streaming port of the camera #1. So yes, I think you've basically just forwarded the streaming URL of the first camera, and the issue should be corrected if you forwarded port 8765 instead.

soapy75, 2022-10-25:
Yes i just worked it out, thanks for the help!

closed 2022-10-25 by soapy75