motioneye-project/motioneye #2622: Motioneye on pi 3 uses 100% on one core with very poor framerates

DuckersMcQuack, 2022-10-24:
I read that gpu hardware accel caused a lockup in 2019, is this still a thing?

How can i enable multicore support to have the other 3 cores do something as well?
If gpu accel has been fixed, how do i enable it?

Pi 3 B+

zagrim, 2022-10-26:
> I read that gpu hardware accel caused a lockup in 2019, is this still a thing?

Are you referring to h264_omx with hw acceleration, or something else? There are instructions (which do require some level of technical understanding and expertise, I admit) on how to apply patches that circumvent the lock-up and re-enable omx in #930.

I don't know if there exists such that you are asking to regarding multicore support. From Motion docs I only found mention of "multiple cores" related to [movie_extpipe_use](https://motion-project.github.io/motion_config.html#movie_extpipe_use) which relates to having a custom process for encoding the saved video. To make use of that, you'd need to also use `movie_extpipe` to configure the command line to `ffmpeg` and put both of these config options in "Extra Motion Options" in ME. But still I would assume that unless `ffmpeg` itself has multicore support, encoding a single file might still be done on just one CPU core (although this way it would be not the same `motion` process runs).

JamBalaya56562, 2026-11-11:
Closing, as zagrim answered this. Today the OMX encoder no longer exists in current Raspberry Pi OS; where a hardware encoder is available, motionEye offers H.264/V4L2M2M as the movie format (`main.html:798`).

closed 2026-11-11 by JamBalaya56562