motioneye-project/motioneye #2623: Video flickers

apla2016, 2022-10-24:
Hello, 
I have freshly loaded (after crash) motionEye Version | 0.42.1; Motion Version | 4.3.2; OS Version | Ubuntu 22.04 running on i3 tiny PC. All freshly installed. Three cameras are connected but all 3 video are flickering regardless of resolution and frame rate. 

Prior to crash same hardware /cameras were working fine with older versions of Ubuntu/motionEye/Motion

TheBlackSheep, 2022-10-24:
Hi - I have the same issues on an i5 with 8Gb running Ubuntu 22.04LTS - I've run the stock version that motioneye installed via the package manager and also built motion from source and re-installed the cameras but it's exactly the same - flickering on all images.  

VLC and ffplay to the same cameras all work fine (i.e. no flickering) as does an older version of motionEye v0.42.1, motion v4.2.2, ubuntu 20.04 which is what I'm trying to replace.

apla2016, 2022-10-26:
> Hi - I have the same issues on an i5 with 8Gb running Ubuntu 22.04LTS - I've run the stock version that motioneye installed via the package manager and also built motion from source and re-installed [...]

Any luck with downgrading Linux?

messenjah, 2022-11-01:
Same for me on fresh Ubuntu 22.04 installation and following the [install instruction](https://github.com/motioneye-project/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29)

EDIT: Just noticed there is even the comment in the instruction which I just overread: ...Ubuntu 22.04 is currently broken (uses a broken version of libmicrohttpd12 0.9.75-3)....

Installed on Ubuntu 20.04 and all is fine

apla2016, 2022-11-05:
Downgrading Ubuntu to 20.04 and reloading MotionEye solved all problems

rimroot, 2023-07-06:
Any chance to fix this on Ubuntu 22.04?

zagrim, 2023-07-07:
Installing dev branch using [these instructions](https://github.com/motioneye-project/motioneye/tree/dev#installation) should be one way to fix the issue (ME 0.43 contains a workaround), and the other solution is to downgrade libmicrohttpd (which you have currently something like 0.9.75 I assume) to 0.9.72 which works ok. The library downgrade option has been discussed e.g. here: https://github.com/motioneye-project/motioneye/issues/2287#issuecomment-1024935426

JamBalaya56562, 2026-11-07:
Closing, as the flicker came from libmicrohttpd 0.9.75, which Motion uses, as zagrim explained above: downgrading it to 0.9.72 fixed it, as did the dev branch. Checked on Debian 13 with libmicrohttpd 1.0.1 and Motion 4.7.1: all 30 live pictures from motionEye 0.45.0a1 came through whole. Please reopen if it still flickers with a current release.

closed 2026-11-07 by JamBalaya56562